cyberman54/ESP32-Paxcounter #540: Lily TTGO V2.0 stopped payload to TTN router after 1st counter

asavvin, 2020-01-31:
Good Day
I have uploaded Lily TTGO V2 with code.
TTN router I am getting following screen:
[image]

log from 

ts_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.88, runmode 0
[I][main.cpp:148] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
[I][main.cpp:150] setup(): Internal Total heap 242880, internal Free Heap 216712
[I][main.cpp:156] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2.3-14-gd3e562907
[I][main.cpp:158] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:160] setup(): Wifi/BT software coexist version 1.1.7
[I][main.cpp:164] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:169] setup(): Arduino LMIC version 3.0.99.10
[I][macsniff.cpp:36] printKey(): DevEUI: xxx
[I][macsniff.cpp:36] printKey(): AppEUI: xxx
[I][macsniff.cpp:36] printKey(): AppKey: xxx
[I][configmanager.cpp:190] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:55] open_storage(): Opening NVS
[I][configmanager.cpp:60] open_storage(): Done
[I][configmanager.cpp:204] loadConfig(): NVRAM settings version = 1.9.88
[I][configmanager.cpp:222] loadConfig(): bsecstate = 0
[I][configmanager.cpp:227] loadConfig(): loradr = 5
[I][configmanager.cpp:235] loadConfig(): txpower = 14
[I][configmanager.cpp:243] loadConfig(): adrmode = 1
[I][configmanager.cpp:251] loadConfig(): screensaver = 0
[I][configmanager.cpp:259] loadConfig(): screenon = 1
[I][configmanager.cpp:267] loadConfig(): countermode = 0
[I][configmanager.cpp:275] loadConfig(): sendcycle = 30
[I][configmanager.cpp:283] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:291] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:299] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:307] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:315] loadConfig(): blescantime = 8
[I][configmanager.cpp:323] loadConfig(): BLEscanmode = 1
[I][configmanager.cpp:331] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:339] loadConfig(): rssilimit = 0
[I][configmanager.cpp:347] loadConfig(): payloadmask = 127
[I][configmanager.cpp:355] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:362] loadConfig(): Done
[I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:54] i2c_scan(): 0x3C: SSD1306 Display controller
[I][i2c.cpp:81] i2c_scan(): I2C scan done, 1 devices found.
[I][main.cpp:279] setup(): Starting Bluetooth...
[I][blecsan.cpp:239] start_BLEscan(): Initializing bluetooth scanner ...
[I][blecsan.cpp:204] register_ble_callback(): Register GAP callback
[I][blecsan.cpp:228] register_ble_callback(): Set GAP scan parameters
[I][blecsan.cpp:252] start_BLEscan(): Bluetooth scanner started
[I][lorawan.cpp:303] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:306] lora_stack_init(): Starting LMIC...
RXMODE_RSSI
681915: engineUpdate, opmode=0x4
682012: TXMODE, freq=904700000, len=23, SF=10, BW=125, CR=4/5, IH=0
[I][main.cpp:370] setup(): Starting Wifi...
I (13007) wifi: wifi driver task: 3ffeac64, prio:23, stack:3584, core=0
I (26212) wifi: wifi firmware version: 7997e4b
I (26213) wifi: config NVS flash: disabled
I (26215) wifi: config nano formating: disabled
I (26219) wifi: Init dynamic tx buffer num: 32
I (26223) wifi: Init data frame dynamic rx buffer num: 32
I (26229) wifi: Init management frame dynamic rx buffer num: 32
I (26234) wifi: Init management short buffer num: 32
I (26239) wifi: Init static rx buffer size: 1600
I (26244) wifi: Init static rx buffer num: 16
I (26247) wifi: Init dynamic rx buffer num: 32
I (26252) wifi: set country: cc=US␀ schan=1 nchan=13 policy=1

W (26257) wifi: ap start fail

I (26260) wifi: Set ps type: 0

I (26274) wifi: mode : null
I (26274) wifi: ic_enable_sniffer
[I][main.cpp:383] setup(): Starting Interrupt Handler...
[I][main.cpp:407] setup(): Starting Timers...
[I][main.cpp:465] setup(): Starting Timekeeper...
[I][timekeeper.cpp:150] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][timesync.cpp:41] send_timesync_req(): [13.089] Timeserver sync request started
[I][main.cpp:472] setup(): Features: OLED OTA BLE LORA FILTER PACKED WIFI
1017025: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 7
1017664: RXMODE_SINGLE, freq=925700000, SF=10, BW=500, CR=4/5, IH=0
[I][lorawan.cpp:104] lora_setupForNetwork(): DEVaddr: 26022B60
[I][lorawan.cpp:108] lora_setupForNetwork(): Radio parameters: SF10 / BW125 / CR 4/5
1023315: engineUpdate, opmode=0x800
1193596: engineUpdate, opmode=0x808
1193709: TXMODE, freq=904300000, len=14, SF=10, BW=125, CR=4/5, IH=0
[I][lorawan.cpp:271] lora_send(): 1 byte(s) sent to LORA
1273524: setupRx1 txrxFlags 0x1 --> 01
start single rx: now-rxtime: 2
1274154: RXMODE_SINGLE, freq=924500000, SF=10, BW=500, CR=4/5, IH=0
1280440: decodeFrame txrxFlags 0x1 --> 21
1280471: Received downlink, window=RX1, port=-1, ack=0, txrxFlags=0x21
[I][lorawan.cpp:519] myRxCallback(): 18 byte(s) downlink MAC commands
[I][lorawan.cpp:527] myRxCallback(): 4 byte(s) uplink MAC commands
1281158: engineUpdate, opmode=0x910
1311211: engineUpdate, opmode=0x110
1311308: TXMODE, freq=903900000, len=16, SF=9, BW=125, CR=4/5, IH=0
1383462: setupRx1 txrxFlags 0x21 --> 01
start single rx: now-rxtime: 1
1384082: RXMODE_SINGLE, freq=923300000, SF=9, BW=500, CR=4/5, IH=0
[I][senddata.cpp:103] sendData(): Counter cleared
4569596: engineUpdate, opmode=0x888
[I][lorawan.cpp:271] lora_send(): 4 byte(s) sent to LORA
[I][senddata.cpp:103] sendData(): Counter cleared

Thank you

cyberman54, 2020-01-31:
Check if you have payload format filters in your TTN console installed for you app.
If so, delete both (decoder + converter) and re-check if you see the raw data.
If yes, reinstall payload converters from repository, this will fix the issue.
If not, report here.

asavvin, 2020-02-01:
Thank you for your comments. 
Still the same TTN stopped after 1st counter with the same message. 
Even if removed payload decoders and converters.
[image]

cyberman54, 2020-02-01:
Did you wait 1 minute after startup? Counter values are sent cyclic every 60 seconds (default).

asavvin, 2020-02-01:
Yes I did. I removed decoder and converter on TTN.
This is the screen TTN Router  I am getting
[image]
and log:
[image]

cyberman54, 2020-02-02:
please post your platformio.ini, paxcounter.conf, and hal .h file, which were used for build.

asavvin, 2020-02-02:
Added ( I renamed with.txt at the end)
Thank you 

[paxcounter.conf.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/4144206/paxcounter.conf.txt)
[platformio.ini.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/4144207/platformio.ini.txt)
[ttgov2.h.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/4144208/ttgov2.h.txt)

cyberman54, 2020-02-02:
Which country / frequency band did you select in `lmic_config.h` ?

Please try disabling interrupts by setting `//#define LMIC_USE_INTERRUPTS 1` in `lmic.config.h`

cyberman54, 2020-02-02:
The log shows that the device is trying to send data, but the LORA stack is busy for some reason. The data is enqueued then to postpone the transmission. Watch the display, you should see the number of enqueued messages in the bottom line on the right. If it increases every minute, messages are enqueued.

We have to find the reason, why your LORA stack is busy.

asavvin, 2020-02-02:
Which country / frequency band did you select in lmic_config.h ? 
North America 915
Please try disabling interrupts by setting //#define LMIC_USE_INTERRUPTS 1 in lmic.config.h
I did it at first time because when it was enabled I was not able to see on TTN router any connection

Please post serial log with debug_level = 5 set in platformio.ini
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): 5d:8e:67:da:2c:31
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -81
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): 62:fb:f2:86:55:3b
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -70
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): 47:a5:ad:66:53:78
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -55
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): 7f:a2:0c:74:4c:41
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -47
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): f4:d3:83:4f:83:b4
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -81
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x1b -> LE Bluetooth Device Address
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): d0:03:df:9f:93:a7
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_PUBLIC
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -85
[V][macsniff.cpp:124] mac_add(): known BLTH RSSI -85dBi -> salted MAC A7941AD0 -> Hash 1E38 -> WiFi:0  BLTH:7 -> 63000 Bytes left
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): 2c:8c:f4:d2:c7:89
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -77
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): dc:56:e7:4d:e3:5c
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_PUBLIC
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -78
[V][macsniff.cpp:124] mac_add(): known BLTH RSSI -78dBi -> salted MAC 5CE3C8D8 -> Hash 1FF7 -> WiFi:0  BLTH:7 -> 62952 Bytes left
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:137] gap_callback_handE (24983) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (24983) task_wdt:  - IDLE0 (CPU 0)
E (24983) task_wdt: Tasks currently running:
E (24983) task_wdt: CPU 0: Btc_task
E (24983) task_wdt: CPU 1: IDLE1
E (24983) task_wdt: Aborting.
abort() was called at PC 0x400e8cbb on core 0

Backtrace: 0x4009364c:0x3ffbe320 0x4009387d:0x3ffbe340 0x400e8cbb:0x3ffbe360 0x40085781:0x3ffbe380 0x40009217:0x3ffd7340 0x40007d13:0x3ffd7360 0x40007c69:0x3ffd7380 0x40008106:0x3ffd73a0 0x400e6ff3:0x3ffd7430 0x400811d2:0x3ffd7490 0x40165f45:0x3ffd74c0 0x40161032:0x3ffd74e0 0x4008f6d1:0x3ffd7510

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.88, runmode 1
[I][configmanager.cpp:190] loadConfig(): Reading settings from NVS
[... 903 lines cut ...]
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -47
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x02 -> Incomplete List of 16-bit Service Class UUIDs
[... cut ...]
[V][blecsan.cpp:153] gap_callback_handler(): BT device filtered
[V][blecsan.cpp:116] gap_callback_handler(): BT payload rcvd -> type: 0x1d -> Simple Pairing Hash C-256
[V][blecsan.cpp:137] gap_callback_handler(): Device address (bda): ec:5c:68:5e:04:2e
[V][blecsan.cpp:139] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_PUBLIC
[V][blecsan.cpp:140] gap_callback_handler(): RSSI                : -63
[V][blecsan.cpp:116] gap_callback_hand

I think with level  5 debug  I stopped getting connection to TTN router.

cyberman54, 2020-02-02:
(Log deleted, since it contained MAC addresses)

cyberman54, 2020-02-02:
Don't know what's going wrong here. Can you re-try with other hardware, but at the same location (Lora gateway)?

asavvin, 2020-02-02:
Ok Thank you I will try.

asavvin, 2020-02-03:
Hi, I was able to get connection and payload with lopy4 ( pycom) 
Thank you
Look like TTGO V2.0 hardware issue.

closed 2020-02-03 by asavvin