cyberman54/ESP32-Paxcounter #554: HELTEC LORA 32 V2 - OTAA join fails - 2

freeedom67, 2020-02-13:
Hello Cyberman,

this issue is related to this issue : https://github.com/cyberman54/ESP32-Paxcounter/issues/411

I have the same Problems with the HELTEC LORA 32 V2 board. I tried both fixes in this thread but unfortunatly they didnt work for me. I tried the latest Version and Version 1.7.93. I only get the Activation Frames in TTN. It just worked once correctlym the other times i just get some time sync
errors in the Serial Console after LMIC-Joined callback.

Thank you for you help!

Greetings Tim

cyberman54, 2020-02-13:
I currently don't have a solution for this. It is yet unclear what's the root cause for this. It probably is a timing issue either in the LMIC stack running on top of paxcounter app, or it's on TTN server site.

To figure this out it will be necessary to have exact timing measures on RF radio level. Look [here](https://github.com/mcci-catena/arduino-lmic/issues/519) for further details. Any help with this issue is highly welcome.

cyberman54, 2020-03-17:
@freeedom67 any news here?

freeedom67, 2020-03-20:
@cyberman54 Unfortunately not, I tried your timing-error-compensation settings again and sometimes it connects properly to TTN- but it cant be reproduced. I think we are going to try different boards next week! Greetings

cyberman54, 2020-03-25:
This should be solved with latest MCCI LMIC release 3.1.0.
I tested on a Heltec V2 board: it now joins immediately, tested on TTN and local Loraserver/Chirpstack.

closed 2020-03-25 by cyberman54