cyberman54/ESP32-Paxcounter #555: ESP32-Paxcounter on Odroid-GO

dl8bb, 2020-02-13:
Hi Cyberman,
  I've tried to compile the paxcounter for [odroid-go](https://wiki.odroid.com/odroid_go/odroid_go) based on m5stack.h and m5fire.h/m5core.h. Some things are working and the device seams to do scanns but no display on the TFT.

```
[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 2748 bytes left | Taskstate = 0
[D][cyclic.cpp:61] doHousekeeping(): LEDloop 492 bytes left | Taskstate = 2
[I][cyclic.cpp:70] doHousekeeping(): Battery: 4154mV
[V][macsniff.cpp:124] mac_add(): new   WiFi RSSI -32dBi -> salted MAC F8084B47 -> Hash 6BE3 -> WiFi:1  BLTH:0 -> 4194216 Bytes left
[V][macsniff.cpp:124] mac_add(): known WiFi RSSI -32dBi -> salted MAC F8084B47 -> Hash 6BE3 -> WiFi:1  BLTH:0 -> 4194216 Bytes left
[V][macsniff.cpp:124] mac_add(): known WiFi RSSI -42dBi -> salted MAC F8084B47 -> Hash 6BE3 -> WiFi:1  BLTH:0 -> 4194216 Bytes left
[V][macsniff.cpp:124] mac_add(): known WiFi RSSI -43dBi -> salted MAC F8084B47 -> Hash 6BE3 -> WiFi:1  BLTH:0 -> 4194216 Bytes left
```

My definitions for the display is driven from the odroid-go wiki

```
// Pins for interface of LC Display
#define MY_OLED_CS GPIO_NUM_5
#define MY_OLED_DC GPIO_NUM_21
#define MY_OLED_CLK GPIO_NUM_18
#define MY_OLED_RST NOT_A_PIN // GPIO_NUM_33
#define MY_OLED_BL GPIO_NUM_14
#define MY_OLED_MOSI GPIO_NUM_23
#define MY_OLED_MISO GPIO_NUM_19
```
How do I get output?

Kind regards,
birko

cyberman54, 2020-02-13:
TFT display control is not implemented yet in display.cpp.
The LC-Display settings in m5core.h/m5fire.h are for future implementation, currently they have no impact.

Maybe you can improve this by your contribution?

dl8bb, 2020-02-14:
Will have a look at it.

cyberman54, 2020-02-14:
@dl8bb please look at this lib, it has similar API as ss_oled and is small and fast:

https://github.com/bitbank2/bb_spi_lcd

cyberman54, 2020-03-17:
@dl8bb any news here? Meanwhile i tested the code on M5stack Fire with Lora868 board, it works.

cyberman54, 2020-04-13:
TFT display support is now implemented (v1.9.99), so i'm closing this.

closed 2020-04-13 by cyberman54