cyberman54/ESP32-Paxcounter #573: OLED not working anymore (TTGOv1)

nuthub, 2020-03-16:
OLED is not recognized with latest version from development branch anymore (worked with a previous version without any issues). Serial console shows the following in a loop:

```
assertion "rc != OLED_NOT_FOUND" failed: file "src/display.cpp", line 57, function: void setup_display(int)
abort() was called at PC 0x4012e40f on core 1

Backtrace: 0x40092798:0x3ffcd6e0 0x400929c9:0x3ffcd700 0x4012e40f:0x3ffcd720 0x400d369d:0x3ffcd750 0x400d3b84:0x3ffcd780 0x400d4bdf:0x3ffcd7c0 0x400e5847:0x3ffcd860 0x4008e8f5:0x3ffcd880

Rebooting...
```

Perhaps it is related to recent update of ss_oled library?

cyberman54, 2020-03-16:
Try to manually delete the precompiled ss_oled lib in platformio.
In pio filesystem look at .pio -> libdeps -> usb -> ss_oled
Delete it, then recompile software.

cyberman54, 2020-03-16:
PS: No, it's not because of the latest ss_oled lib update (4.0.1 / 4.1.0).
The version in dev branch already has the necessary changes.

nuthub, 2020-03-16:
I already tried after removing complete .pio (and .vscode) directory.
Tested also with latest commit of dev branch.

cyberman54, 2020-03-17:
I don't have a TTGOv1 for testing. Tested on a TTGOv21, this one works.

cyberman54, 2020-03-17:
@nuthub Could you please check with serial console if the OLED display is detected as I2C device during startup? You should see this:

```
[I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:54] i2c_scan(): 0x3C: SSD1306 Display controller
```

If this line is missing, the display was not detected an we have to investigate, why.

nuthub, 2020-03-17:
After fixing (not sure if correctly fixed) ss_oled library (see https://github.com/bitbank2/ss_oled/pull/26) and doing the modifications in PR #574 my OLED is working again.

nuthub, 2020-03-17:
> @nuthub Could you please check with serial console if the OLED display is detected as I2C device during startup? You should see this:
>
> ```
> [I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan... [...]

After implementing my fix described above, I can see these lines in the serial console.

cyberman54, 2020-03-18:
Fixed.

closed 2020-03-18 by cyberman54