cyberman54/ESP32-Paxcounter #575: SDS011: anything missing?

AugustQu, 2020-03-18:
Hi,

I just made another test wirh my hardware and software here for PAYLOAD_ENCODER set to 3 (=Cayenne LPP dynamic). In the TTN-console I received these values:
`156500012065000021650000`

In the console this converts to:
`{
  "luminosity_21": 1,
  "luminosity_32": 0,
  "luminosity_33": 0
}
`

To me this seems to be correct.

So my question is: what is missing?

I think I have to write some lines for the README. OK, will do.

What else?

August

cyberman54, 2020-03-18:
@AugustQu I think this should work, and SDS011 is done.
You could validate by creating an account at mydevices.com, then use the Cayenne App to check if it shows the data.

AugustQu, 2020-03-20:
Hi,

today I could validate it: in mydevices I can see the data for smartphone, PM10 and PM2.5.

So everything seems to be fine.

cyberman54, 2020-03-20:
Perfect! So we close this here. Thank you for all the efforts and contributing to this project!

closed 2020-03-20 by cyberman54