cyberman54/ESP32-Paxcounter #576: ttgo2.1.6 mit bme680 und NEU sds011 sendet keine daten ins ttn

jensileinchen, 2020-03-20:
hi cyberman -- ich habe heute die neuste software auf meinen paxcounter gespielt, hal etc entsprechend angepasst, payload encoder / decoder im ttn angepasst aber ich bekomme keine lesbaren daten mehr vom paxcounter -- die vorlagen habe ich aus dem generic geholt und in meine srx/hal ttgo21new eingefügt -- 
beim powerup läuft der sds einmal an, macht aber danach nix mehr - der pax zählt brav, sendet nix ins ttn, der sds (vorher klassisch in nem feinstaubsensor vom okLAB stuttgart mit nodeMCU eingebaut) macht nach einmaligen starten nix mehr ....

kann ich irgendwas noch testen -- ich befürchte mit meiner beschreibung ist das eher wie stochern im heuhaufen nach der nadel .....

cyberman54, 2020-03-20:
Bitte hier posten:
- hal File
- serial console log
- paxcounter.conf

x-RoNiN-x, 2020-03-20:
@jensileinchen  ich poste dies mal hier!

hal File

`// clang-format off
// upload_speed 921600
// board esp32dev

#ifndef _TTGOV21NEW_H
#define _TTGOV21NEW_H

#include <stdint.h>

/*  Hardware related definitions for TTGO V2.1 Board
// ATTENTION: check your board version!
// This settings are for boards labeled v1.6 on pcb, NOT for v1.5 or older
*/

#define HAS_LORA 1       // comment out if device shall not send data via LoRa
#define CFG_sx1276_radio 1 // HPD13A LoRa SoC

// enable only if you want to store a local paxcount table on the device
#define HAS_SDCARD  1      // this board has an SD-card-reader/writer
// Pins for SD-card
#define SDCARD_CS    (13)
#define SDCARD_MOSI  (15)
#define SDCARD_MISO  (2)
#define SDCARD_SCLK  (14)

// BME280 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME280 GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL
#define BME280_ADDR 0x76 // change to 0x77 depending on your wiring

// SDS011 dust sensor settings
#define HAS_SDS011 1 // use SDS011
// used pins on the ESP-side:
#define SDS_TX 19     // connect to RX on the SDS011
#define SDS_RX 23     // connect to TX on the SDS011

#define HAS_DISPLAY 1
#define HAS_LED (25) // green on board LED
#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

// Pins for I2C interface of OLED Display
#define MY_OLED_SDA (21)
#define MY_OLED_SCL (22)
#define MY_OLED_RST NOT_A_PIN

// Pins for LORA chip SPI interface, reset line and interrupt lines
#define LORA_SCK  (5) 
#define LORA_CS   (18)
#define LORA_MISO (19)
#define LORA_MOSI (27)
#define LORA_RST  (23)
#define LORA_IRQ  (26)
#define LORA_IO1  (33)
#define LORA_IO2  (32)

#endif
`

paxcounter.conf

`// clang-format off

// ----- Paxcounter user config file ------
//
// --> adapt to your needs and use case <--
[... 88 lines cut ...]
#define CONFIGPORT                      3       // config query results
#define GPSPORT                         0       // gps - set to 1 to send combined GPS+COUNTERPORT payload  
#define BUTTONPORT                      5       // button pressed signal
#define BEACONPORT                      6       // beacon alarms
#define BMEPORT                         7       // BME680 sensor
#define BATTPORT                        8       // battery voltage
#define TIMEPORT                        9       // time query and response
#define TIMEDIFFPORT                    13      // time adjust diff
#define SENSOR1PORT                     10      // user sensor #1
#define SENSOR2PORT                     11      // user sensor #2
#define SENSOR3PORT                     12      // user sensor #3

// Cayenne LPP Ports, see https://community.mydevices.com/t/cayenne-lpp-2-0/7510
#define CAYENNE_LPP1                    1       // dynamic sensor payload (LPP 1.0)
#define CAYENNE_LPP2                    2       // packed sensor payload (LPP 2.0)
#define CAYENNE_GPS                     3       // full scale GPS payload
#define CAYENNE_ACTUATOR                10	    // actuator commands
#define CAYENNE_DEVICECONFIG            11	    // device period configuration
#define CAYENNE_SENSORREAD              13	    // sensor period configuration
#define CAYENNE_SENSORENABLE            14	    // sensor enable configuration
`

welchen log meinst du`?

wir schubsen die daten mittels VSCode hoch

cyberman54, 2020-03-20:
We need serial log written during runtime here.

cyberman54, 2020-03-20:
@AugustQu ping

AugustQu, 2020-03-21:
does it send no data to TTN at all?

In the last days I checked my code here (identical to the code here, branch SDS011). In 3 days  I had:

- connected to TTN and send values

- no connection, no data sent

- connected to TTN and sent data

So: does it send data without SDS011?

cyberman54, 2020-03-21:
Check if the device is joined. Watch out for the join notice in serial log, or the LED flashing fast.
No join, no data.

jensileinchen, 2020-03-21:
///// SERIAL LOG MONITOR
actual setup is paxcounter 2.1.6 "ttgov2.1new" only with sds011 connected on GND +5VDC RX TX

> Executing task: C:\Users\jens_\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --- 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:000,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.98, runmode 0
[I][configmanager.cpp:190] loadConfig(): Reading settings from NVS      
[I][configmanager.cpp:55] open_storage(): Opening NVS
[I][configmanager.cpp:60] open_storage(): Done
[I][configmanager.cpp:204] loadConfig(): NVRAM settings version = 1.9.98
[I][configmanager.cpp:222] loadConfig(): bsecstate = 139
[I][configmanager.cpp:227] loadConfig(): loradr = 5
[I][configmanager.cpp:235] loadConfig(): txpower = 14
[I][configmanager.cpp:243] loadConfig(): adrmode = 1       
[I][configmanager.cpp:251] loadConfig(): screensaver = 0
[I][configmanager.cpp:259] loadConfig(): screenon = 1
[I][configmanager.cpp:267] loadConfig(): countermode = 0
[I][configmanager.cpp:275] loadConfig(): sendcycle = 30
[I][configmanager.cpp:283] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:291] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:299] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:307] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:315] loadConfig(): blescantime = 8
[I][configmanager.cpp:323] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:331] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:339] loadConfig(): rssilimit = 0
[I][configmanager.cpp:347] loadConfig(): payloadmask = 127
[I][configmanager.cpp:355] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:362] loadConfig(): Done
[I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:54] i2c_scan(): 0x3C: SSD1306 Display controller
[I][i2c.cpp:81] i2c_scan(): I2C scan done, 1 devices found.
[I][main.cpp:242] setup(): Starting LED Controller...
[I][power.cpp:171] calibrate_voltage(): ADC characterization based on reference voltage stored in eFuse
[I][lorawan.cpp:297] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:300] lora_stack_init(): Starting LMIC...
[I][main.cpp:331] setup(): init fine-dust-sensor
[I][main.cpp:375] setup(): Starting Wifi...
[I][main.cpp:388] setup(): Starting Interrupt Handler...
[I][main.cpp:412] setup(): Starting Timers...
[I][main.cpp:470] setup(): Starting Timekeeper...
[I][timekeeper.cpp:151] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][timesync.cpp:52] timesync_request(): [10.519] Timeserver sync request seqNo#220 started
[I][main.cpp:477] setup(): Features: OLED LED BATT OTA LORA SDS PACKED WIFI
[I][lorawan.cpp:79] lora_setupForNetwork(): DEVaddr: 0x26012140 | Network ID: 0x000013 | Network Type: 0
[I][lorawan.cpp:80] lora_setupForNetwork(): RSSI: 27 | SNR: 7
[I][lorawan.cpp:84] lora_setupForNetwork(): Radio parameters: SF7 | BW125 | CR 4/5
[W][timesync.cpp:234] timesync_serverAnswer(): [21.604] Network did not answer time request
[W][timesync.cpp:106] timesync_processReq(): [21.604] Timesync aborted: handshake out of sync
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4160mV
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4204mV
[I][sds011read.cpp:35] sds011_loop(): SDS011 error: 1
[I][senddata.cpp:110] sendData(): Counter cleared
[I][lorawan.cpp:265] lora_send(): 6 byte(s) sent to LORA
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4284mV
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4478mV
[I][sds011read.cpp:35] sds011_loop(): SDS011 error: 1
[I][senddata.cpp:110] sendData(): Counter cleared
[I][lorawan.cpp:265] lora_send(): 6 byte(s) sent to LORA
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4344mV
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4234mV 
[I][sds011read.cpp:35] sds011_loop(): SDS011 error: 1
[I][senddata.cpp:110] sendData(): Counter cleared       
[I][lorawan.cpp:265] lora_send(): 6 byte(s) sent to LORA

jensileinchen, 2020-03-21:
its connected to ttn and i can see payload in ttn-network console / devices  --- but not as readable decoded payload ..

AugustQu, 2020-03-21:
ich werde nachher noch mal mein Teil hier testen. Ich habe mich bisher damit zufrieden gegeben, wenn ich Daten in der TTN-Console in hex gesehen habe.

Ah, ich sehe:
`[I][sds011read.cpp:35] sds011_loop(): SDS011 error: 1`

Irgendetwas klappt in der Kommunikation nicht. Läuft der Sensor überhaupt (hörbar)?

x-RoNiN-x, 2020-03-21:
yes the SDS011 starts when the whole setup is switched on, but does not stop!

Maybe we made a mistake when compiling the code!
We use the Paxcounter ttgo2,1new.

Do you have to add anything else from the Generic.h to the board file?
We inserted the block with the SDS011.

cyberman54, 2020-03-21:
Your wiring / pin mapping is wrong, using same pins for serial and SPI.
This can't work.

> #define SDS_TX 19 // connect to RX on the SDS011
> #define SDS_RX 23 // connect to TX on the SDS011
> ...
> #define LORA_MISO (19)
> #define LORA_RST (23)

AugustQu, 2020-03-21:
@cyberman54 
I have the same settings here. I will check it.

AugustQu, 2020-03-21:
@jensileinchen 
in the file sds011read.cpp can you please outcomment the calls to sleep and wakeup, e.g.:
`void sds011_sleep(void) {
  ESP_LOGI(TAG, "SDS011: set to sleep");
  //sdsSensor.sleep();
  isSDS011Active = false;
}
`

cyberman54, 2020-03-21:
> @cyberman54
> I have the same settings here. I will check it.

Maybe your device has no LORA chip?

x-RoNiN-x, 2020-03-22:
Hello....

The wiring is now correct. I change the pins in the board config.
Now the TTN Paxcounter receive data from SDS011 but the sensor doesn´t sleep! I have tried to commend it out, no way! furthermore sends the sds over port 1 and make a misstake in paxes. i tried to change the port but i failed.

and yes, our device has a lora chip.

thanks and a good day 

greets

AugustQu, 2020-03-22:
here is the pinout of the board: https://ae01.alicdn.com/kf/H21444c9d0105413fa522d620d7a5cddaR.jpg

which pins did yopu use?

wakeup/sleeping of the sensor is handled in _sds011read.cpp_, functions _sds011_wakeup_() and _sds011_sleep_(). In these functions simply the sensor is called. Are these calls commented out?

I checked the workings of this in a tiny piece of code which did nothing else than putting the sensor to sleep, wait 10 seconds, wakeup and wait again 10 seconds - and so on. This worked here so I copied the code to the PaxCounter-code and didn't look at this topic again (I had much more problems with the LoRaWAN-connection).

x-RoNiN-x, 2020-03-22:
I use the pins PIO34 and PIO35.

The sds011_wakeup() and sds011_sleep() is uncommented. 
Our Setup sends into the TTN. I encrypt the Hex and put the data in a grafana overview.
But the SDS dont sleep. it is not so good for battery lifetime.

greets
and thanks

AugustQu, 2020-03-23:
@x-RoNiN-x 
I try to find the reason for this.

Currently I can only confirm this behaviour.

x-RoNiN-x, 2020-03-30:
Hi

We now have a configuration that works for us.
The Paxcounter sends Pax Temp Hum Pres Air every minute. The Sds sends its data every 5 minutes, to the TTN.
We test this configuration a little longer!

AugustQu, 2020-03-30:
@x-RoNiN-x 

what did you change in the code/configuration?

AugustQu, 2020-03-30:
I played with different configurations:
- different pins
- different HardwareSerials
- even a SoftwareSerial
did not work as intended.

And I had the effect that sleep() worked, but not wakeup()

I am playing with a minimal program to test this behaviour. This tiny little piece of code works on a Arduino-board, but not with the ESP32....

x-RoNiN-x, 2020-03-31:
Hi,
I tested it longer yesterday. There is a problem that you seem to have.
The SDS has an initialization time of 30 seconds as soon as it is connected to power. Then the time that the paxcounter sends to the ttn is no longer correct.
So switch everything on until the fan stops rotating and carry out a soft reset at this moment.
The problem can be solved in the code. But I'm not a born programmer :-D

x-RoNiN-x, 2020-03-31:
I use the SDS with a Paxcounter v 1.6. In the hal folder it is the ttgov21new.h.

// SDS011 dust sensor settings
#define HAS_SDS011 1 // use SDS011
// used pins on the ESP-side:
#define SDS_TX 12     // connect to RX on the SDS011
#define SDS_RX 35     // connect to TX on the SDS011

x-RoNiN-x, 2020-03-31:
The error will be that the SDS and the Paxcounter run differently. You would have to give the SDS the command to start and stop! See also the .ino from the AirRohr project Stuttgart!
[](https://github.com/goggel/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino) at line 220, 533, 565,

the lines from 1781 are interesting!!!

x-RoNiN-x, 2020-04-11:
Hi there,
we have found a solution to our problem and no longer need any information. thank you for your patience. greets

closed 2020-04-11 by cyberman54