cyberman54/ESP32-Paxcounter #577: TTGO V1 SX127x pin

DierkJ, 2020-03-22:
Hi Cyberman,

first of all: Thanks for the great project!
I found a little issue with the TTGO V1 hardware, which results in a non working LMIC.  The DIO2 pin is wired to pin 32 of the ESP on my TTGO V1 board.  Leaving the pin LMIC_UNUSED_PIN will not wakeup the SX1272.

Exchange
  `#define LORA_IO2  LMIC_UNUSED_PIN`
to
  `#define LORA_IO2  (32) ` 
will solve this.

cyberman54, 2020-03-22:
DIO2 pin ist not used for LORA modulation, only for FSK (timeout signal).
Thus, if you do not use FSK, DIO2 ist not needed.

DierkJ, 2020-03-23:
Sorry for my typo about the SX1272.  My TTGO V1 is using the SX1276. Just double-checked on the pcb.

If the Dio2 is set to unused the first call to txlora() will fail in line 799:
`   opmodeLora();`
`    ASSERT((readReg(RegOpMode) & OPMODE_LORA) != 0);`

Looks like the SX is not switched to Lora.

mabuberg, 2020-03-23:
I have a TTGO V1 using a SX1276 too. It doesn't connect to TTN. It outputs "JOINING" on the display and shortly after "TXSTART" where it stays.
I tried switching LORA_IO2 to (32). Same behavior.

Thank you for your work!

cyberman54, 2020-03-23:
Thank you guys for Investigating this. Could you please
1) open Pull Request for the hal files of boards which have DIO2 wired

2) Open pull request for boards which do not have DIO2 wired (add note that is has to be manually wired)

3) raise an issue in MCCI LMIC repository

?

Thanks.

DierkJ, 2020-03-24:
My TTGO V1 looks like this:
[image]

cyberman54, 2020-03-25:
This is not the "original" TTGO T1, looks like a fake Heltec board.

cyberman54, 2020-03-25:
PR merged.

closed 2020-03-25 by cyberman54