cyberman54/ESP32-Paxcounter #578: core dump heltecV2 on assert oled init

mikethebee, 2020-03-23:
I have tried to install 1.9.98 over a 1.9.91 working version on a HeltecV2 LoRa with oled. It got into a boot loop with "assert .... failed" on the oledInit call in display.cpp. I tried esptool erase_flash but the tool returns "ESP32 ROM does not support function erase_flash" 
I am able to return to a working 1.9.91 version okay. Any Thoughts?

cyberman54, 2020-03-25:
I did some investigation on this and found the root cause: The reset logic of OLED library is causing this issue. I made a [pull request](https://github.com/bitbank2/ss_oled/pull/28) already, so this issue will be solved soon. Meanwhile you can attach the patch by hand in the lib.

mikethebee, 2020-03-25:
Thx, I will follow the patch request over on ss_oled.

closed 2020-03-25 by mikethebee