cyberman54/ESP32-Paxcounter #580: FAILURE on Starting LMIC... [TTGOv1]

neolinux, 2020-03-28:
Hello,

anyone got it to work on ttgov1? 

Oled doesent work (is enabled in the configuration)

and in the console i become this error:

`
[I][lorawan.cpp:297] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:300] lora_stack_init(): Starting LMIC...
FAILURE 
.pio\libdeps\usb\MCCI LoRaWAN LMIC library_ID5774\src\lmic\radio.c:799
`

any help?

Thank You.

neolinux, 2020-03-28:
I have this board

[image]

cyberman54, 2020-03-28:
@DierkJ can you help here? Same board as yours?

cyberman54, 2020-03-28:
@neolinux please try version from development branch and make these patches:

add to / change in `TTGOV1.h`:

> #define MY_OLED_ADDR 0x3D
> #define LORA_IO2  LMIC_UNUSED_PIN

neolinux, 2020-03-28:
Same thing, oled is black (it worked as i became the ttgov1 with the demo firmware) and nothing in TTN/same error in the terminal of VS Code.

neolinux, 2020-03-29:
I've got the lorawan to work:

i changed this:

// enable only if you want to store a local paxcount table on the device
#define HAS_SDCARD  1      // this board has an SD-card-reader/writer
// Pins for SD-card
#define SDCARD_CS    (37)
#define SDCARD_MOSI  (38)
#define SDCARD_MISO  (36)
#define SDCARD_SCLK  (39)

neolinux, 2020-03-29:
For the OLED, what i see in the example sketch of ttgov1:

digitalWrite (16, LOW); // set GPIO16 low to reset OLED
  delay (50);
  digitalWrite (16, HIGH); // while OLED is running, GPIO16 must go high

it is implemented in the ESP32-Paxcounter code?

cyberman54, 2020-03-29:
Why did you enable the SD card function? Your picture is showing a board without SD card slot?

neolinux, 2020-03-29:
It doesn’t compile with define HAS_SDCARD 0

cyberman54, 2020-03-29:
you need to comment the define:
// define HAS_SDCARD 1

cyberman54, 2020-03-29:
try to set on ttgov1.h

#DEFINE MY_DISPLAY_RST 16

neolinux, 2020-03-29:
Ok, i've commented the HAS_SDCARD definition, no compilation error :-P

With the #DEFINE MY_DISPLAY_RST 16 the oled continues to be black, so it doesent work.

cyberman54, 2020-03-29:
please reload code from development branch (v1.9.982), the use ttgov1.h without any modifications. Should work then.

neolinux, 2020-03-29:
It works now!

Thank You ;)

closed 2020-03-29 by cyberman54