cyberman54/ESP32-Paxcounter #583: compile errors in src/display.cpp

robomotic, 2020-04-01:
Hello folks,
very neat project I have just started to use PlatformIO so please apologies if this is obvious.

Board: 
[LILYGO® TTGO LORA V1.3 868Mhz ESP32 Chip SX1276 Module 0.96 Inch OLED Screen WIFI And Bluetooth Development Board](https://www.aliexpress.com/item/4000628100802.html)
Setup:

- OS: Darwin x64 18.7.0
- VSS Code: Version: 1.43.2
- Platform IO: Core 4.3.1 Home 3.1.1

Steps:

1. Clone Git Project from PlatformIO
2. Change platform.ini to: halfile = ttgov21new.h
3. Copy loraconf sample and ota.conf
4. In loraconf enable LORA_ABP
4. Build USB environment

Result:

> Compiling .pio/build/usb/src/led.cpp.o
> src/display.cpp: In function 'void dp_setup(int)':
> src/display.cpp:59:21: error: 'MY_DISPLAY_RST' was not declared in this scope
>                      MY_DISPLAY_RST,
>                      ^
> src/display.cpp: In function 'void dp_shutdown()':
> src/display.cpp:486:25: error: 'oledShutdown' was not declared in this scope
>      oledShutdown(&ssoled);
>                          ^
> *** [.pio/build/usb/src/display.cpp.o] Error 1

Observations:
I am assuming that enabling LORA_ABP should exclude the ota.conf include but this does not happen.

I found that this is defined in ttgov21old.h:

> // Pins for I2C interface of OLED Display
> #define MY_DISPLAY_SDA (21)
> #define MY_DISPLAY_SCL (22)
> #define MY_DISPLAY_RST NOT_A_PIN

However this is not present in ttgov21new.h:

> // Pins for I2C interface of OLED Display
> #define MY_OLED_SDA (21)
> #define MY_OLED_SCL (22)
> #define MY_OLED_RST NOT_A_PIN

I wonder if this is just the case or renaming MY_OLED to MY_DISPLAY?

I have no idea what is the issue with the oledShutdown function but I can see an old issue that mentioned it was fixed.

Cheers.

cyberman54, 2020-04-01:
first problem was, indeed, a missing update on ttgov21new.h -> fixed.
second problem came from ss_oled lib, meanwhile this was updated -> fixed.

Please reload and recompile code, and if this issue is solved, please close it. Thx.

cyberman54, 2020-04-01:
waiting for confirmation.

robomotic, 2020-04-01:
will test shortly.

robomotic, 2020-04-01:
I did a full git clone and saw your changes, it doesn't seem like the lib dependency fix worked:

```
src/display.cpp: In function 'void dp_shutdown()':
src/display.cpp:486:25: error: 'oledShutdown' was not declared in this scope
     oledShutdown(&ssoled);
                         ^
*** [.pio/build/usb/src/display.cpp.o] Error 1

```
The OLED PIN settings are okay now.

cyberman54, 2020-04-01:
Is now fixed. (There was a breaking change in the ss_oled API :-( )

closed 2020-04-01 by cyberman54