cyberman54/ESP32-Paxcounter #584: bogus wifi packet in uplink message

robomotic, 2020-04-02:
Hello!
I am scratching my head around this now.

**Board:**
LILYGO® TTGO LORA V1.3 868Mhz ESP32 Chip SX1276 Module 0.96 Inch OLED Screen WIFI And Bluetooth Development Board

**Setup:**

OS: Darwin x64 18.7.0
VSS Code: Version: 1.43.2
Platform IO: Core 4.3.1 Home 3.1.1
TTN Server: Enterprise
Gateway: Kona Mikro Gateway

**Steps:**

1. Clone Git Project from PlatformIO
2. Change platform.ini to: halfile = ttgov21new.h
3. Copy loraconf sample and ota.conf
4. In loraconf enable LORA_ABP
5. Paxcounter: the default with the change (see below)
6. TTN Application I have configured with the Javascript packed_decoder.js insid the uplink decoder section
7. I can see the data flowing correctly

Paxcounter edits:

> // Verbose enables additional serial debug output
> #define VERBOSE                         1       // set to 0 to silence the device, for mute use build option
> 

**Result:**
The paxcounter is working and on the OLED display I can see my home readings ranging from min = 0 to max = 2 (devices).

However the uplink payloads seems bogus see the following:

> uplink_message": {
>     "f_port": 1,
>     "f_cnt": 94,
>     "frm_payload": "DwA=",
>     "decoded_payload": {
>       "wifi": 15
>     },..
> 
> 
>   "uplink_message": {
>     "f_port": 1,
>     "f_cnt": 95,
>     "frm_payload": "qXo=",
>     "decoded_payload": {
>       "wifi": 31401
>     },...
> 
> 
>   "uplink_message": {
>     "f_port": 1,
>     "f_cnt": 96,
>     "frm_payload": "vHE=",
>     "decoded_payload": {
>       "wifi": 29116
>     },...

Since the settings in paxcounter (as per default) are set to this:

#define COUNTERMODE                     0       // 0=cyclic, 1=cumulative, 2=cyclic confirmed

I would expect the Wifi counter to be either 0,1 or 2 in my case.

It seems instead is doing some cumulative counts or something else weird going on.
Cheers.

cyberman54, 2020-04-03:
Could you please remove the decoder from TTN console, then check the incoming raw data in console.
If raw data is correct, something is wrong with the decoder.

robomotic, 2020-04-03:
Okay so you mean you want to see the byte array field?

cyberman54, 2020-04-03:
Yes, but AFTER you removed the decoder.

robomotic, 2020-04-03:
Perfect I removed the decoder and set to None.
Restarted everything and I am seeing similar payloads, I have obscured some sensitive fields but the main content is here:

> {
>   "@type": "type.googleapis.com/ttn.lorawan.v3.ApplicationUp",
>   "end_device_ids": {
>     "device_id": "paxcounter-1",
>     "application_ids": {
>       "application_id": "paxcounter"
>     },
>     "dev_eui": "xxxxxxxxxx",
>     "dev_addr": "xxxxxx"
>   },
>   "correlation_ids": [
> .....
>   ],
>   "received_at": "2020-04-03T22:37:27.855838502Z",
>   "uplink_message": {
>     "f_port": 1,
>     "f_cnt": 6,
>     "frm_payload": "aUM=",
>     "rx_metadata": [
>       {
>         "gateway_ids": {
>           "gateway_id": "xxxxxx",
>           "eui": "xxxxx"
>         },
>         "timestamp": 1051730332,
>         "rssi": -54,
>         "channel_rssi": -54,
>         "snr": 7.2,
>         "uplink_token": "xxxxxxx",
>         "channel_index": 6
>       }
>     ],
>     "settings": {
>       "data_rate": {
>         "lora": {
>           "bandwidth": 125000,
>           "spreading_factor": 7
>         }
>       },
>       "data_rate_index": 5,
>       "coding_rate": "4/5",
>       "frequency": "868300000",
>       "timestamp": 1051730332
>     },
>     "received_at": "2020-04-03T22:37:27.646274489Z"
>   }
> }
> 

so if I base decode the payload: aUM= -> 69 43 in hex.

Right now I am reading between 3 and 4 counts on the display.

Cheers.

cyberman54, 2020-04-04:
Did you decode the payload in the right way, e.g. using an [online packet decoder](https://www.thethingsnetwork.org/forum/t/online-lorawan-packet-decoder/5095)?

robomotic, 2020-04-04:
Hello,
Yes I did.
if necessary I can also provide the payload dump at the Kona Mircrogateway but I have no idea about how to decode them.

robomotic, 2020-04-04:
Is there a way to enable debug from usb serial to see how the payload that is being sent to the Lora component.
That would be hopeful.
Also should I switch to Plain maybe to check if that works ?
Cheers.

robomotic, 2020-04-07:
Hell guys, I added this simple line on lorawan.cpp:

```
ESP_LOGI(TAG, "%d byte(s) sent to LORA", SendBuffer.MessageSize);
ESP_LOG_BUFFER_HEXDUMP(TAG, SendBuffer.Message, SendBuffer.MessageSize, ESP_LOG_DEBUG);
```

With the plain mode, I can now see the values which seem correct (reading 2 counts in this chunk):

```
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 1744 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 1908 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:64] doHousekeeping(): LEDloop 480 bytes left | Taskstate = 2
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4200mV
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 1744 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 1908 bytes left | Taskstate = 1
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:64] doHousekeeping(): LEDloop 480 bytes left | Taskstate = 2
[I][cyclic.cpp:73] doHousekeeping(): Battery: 4154mV
[I][senddata.cpp:110] sendData(): Counter cleared
[D][lorawan.cpp:461] myEventCallback(): TXSTART         
[I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
D (201146) lora: 0x3ffb0b78   00 02                                             |..|
[D][lorawan.cpp:461] myEventCallback(): RXSTART         
[D][lorawan.cpp:461] myEventCallback(): RXSTART         
[D][lorawan.cpp:461] myEventCallback(): TXCOMPLETE   
```   

I have opened a ticket on the TTN support it seems like to be related a bug in their backend, in particular when you change Uplink Decoder back and forth seems to be screwing their database.

Will update once I know more from them.

Thanks for the patience.

cyberman54, 2020-04-07:
Did you test with TTNv2 or TTNv3 backend?
I've never seen this issue on TTNv2 yet.
May you post this to TTN forum?

robomotic, 2020-04-13:
Hello,
sorry for the delay, I am using the Enterprise version of TTN but the bug was also found on the TTN community edition.
Basically what I discovered (after painful trial and errors) is that with a certain sequence of actions involving setting the payload decoder, the application key is regenerated thus making the payload look random.
I have sent the bug description to support 2 weeks ago but they haven't replied yet .....
Anyway feel free to close this!

cyberman54, 2020-04-13:
I will close this issue now, since it is no paxcounter issue. But please post the solution here, after you got one. Thanks.

closed 2020-04-13 by cyberman54