cyberman54/ESP32-Paxcounter #585: compiler error on `payload.cpp`

ctron, 2020-04-03:
I have a compiler error in `payload.cpp`:

~~~
src/payload.cpp: In member function 'void PayloadConvert::addSDS(sdsStatus_t)':
src/payload.cpp:350:30: error: qualified-id in declaration before '(' token
 void PayloadConvert::addCount(uint16_t value, uint8_t snifftype) {
                              ^
src/payload.cpp:527:1: error: expected '}' at end of input
 }
 ^
~~~

Taking a look at the source, it looks indeed weird, as the function `addSDS` is not completed properly:

https://github.com/cyberman54/ESP32-Paxcounter/blob/886587cacb8428910cc10927658c0fdecb46dd65/src/payload.cpp#L332-L350

cyberman54, 2020-04-03:
@AugustQu can you help here?

cyberman54, 2020-04-03:
should be fixed in master now, could you please reload and test?

AugustQu, 2020-04-03:
sorry for being absent. I will look at it tomorrow.

AugustQu, 2020-04-03:
looking in my code here it says:
`// value of PM2.5
#if (PAYLOAD_ENCODER == 3) // Cayenne LPP dynamic
    buffer[cursor++] = LPP_PARTMATTER25_CHANNEL;    // for PM2.5
#endif
    buffer[cursor++] = LPP_LUMINOSITY; // workaround since cayenne has no data type meter
    buffer[cursor++] = highByte((uint16_t)(sds.pm25 * 10));
    buffer[cursor++] = lowByte((uint16_t)(sds.pm25 * 10));
#endif  // HAS_SDS011
}
`

AugustQu, 2020-04-03:
ups, this is the important part:
#endif  // HAS_SDS011
}

void PayloadConvert::addCount(uint16_t value, uint8_t snifftype) {

AugustQu, 2020-04-04:
I made a download of the master-branch and it compiles and works (but I only checked for the SDS011-part).

One thing that still does not work here is the handling of the wakeup/sleep. I will again check this on monday.

closed 2020-04-05 by cyberman54