cyberman54/ESP32-Paxcounter #586: Can't connect a BME680 to TTGO LORAv1 (with SDS011)

theArcher73, 2020-04-06:
Hello,
I'm trying to run an SDS011 and a BME680 on a TTGO LoRAv1. The SDS is running, on the BME680 I get the following messages in the monitor:

[I][main.cpp:388] setup(): Starting Interrupt Handler...
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
[I][bmesensor.cpp:64] bme_init(): BSEC v1.4.7.4
[E][bmesensor.cpp:158] checkIaqSensorStatus(): BME680 error -2
[E][bmesensor.cpp:71] bme_init(): BME680 sensor not found

The BME is connected to 3.3V and GND, SDA to pin 21 and SCL to pin 22, SDIO is bridged to GND at the BME.

In the HAL file for the board I have implemented the following extensions: 

// enable only if device has these sensors, otherwise comment these lines
// BME680 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME680 SDA, SCL
#define BME680_ADDR BME680_I2C_ADDR_PRIMARY // !! connect SDIO of BME680 to GND !!

Does anyone have a tip for me on how I can get the part to fly/ better yet get readings into the payload? Thanks a lot in advance and many greetings from Saxony

Archer

x-RoNiN-x, 2020-04-06:
Put your SDA on pin 4 and scl on pin 15 and try again. greets

theArcher73, 2020-04-06:
I][i2c.cpp:23] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:71] i2c_scan(): I2C scan done, 0 devices found.
.....
[I][bmesensor.cpp:64] bme_init(): BSEC v1.4.7.4
[E][bmesensor.cpp:158] checkIaqSensorStatus(): BME680 error -2
[E][bmesensor.cpp:71] bme_init(): BME680 sensor not found

The connection SDIO to GND is really necessary? (Didn't I have one of the NODE.MCU...)

Archer

x-RoNiN-x, 2020-04-06:
// BME680 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME680 GPIO_NUM_4, GPIO_NUM_15 // SDA, SCL
#define BME680_ADDR BME680_I2C_ADDR_PRIMARY // connect SDIO of BME680 to GND

Is this correct in the HAL file TTGO_Lorav1?

theArcher73, 2020-04-06:
I have adapted the HAL file, unfortunately without success. No Idea whats to do. It's a little depressing right now.

cyberman54, 2020-04-06:
> connect SDIO of BME680 to GND

Did you do / verify this?

cyberman54, 2020-04-06:
> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0

looks like your I2C bus is not properly working. Check bus signals / voltages and pullup resistors.

> #define BME680_ADDR BME680_I2C_ADDR_PRIMARY

You may try BME680_I2C_ADDR_SECONDARY depending on you hardware (SDIO wiring)

theArcher73, 2020-04-06:
Yes, I did this.
What happens id Not? Will this missing Link destroy the Sensor or appears only an error?

theArcher73, 2020-04-06:
> > [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
> 
> 
> 
> looks like your I2C bus is not properly working. Check bus signals / voltages and pullup resistors.

I can check the resistors. The Bus-Signal, is ist possible with only a Multimeter?
> 
> 
> 
> > #define BME680_ADDR BME680_I2C_ADDR_PRIMARY
> 
> 
> 
> You may try BME680_I2C_ADDR_SECONDARY depending on you hardware (SDIO wiring)

I will do this, tomorrow. Thx dir your Support, maybee tomorrow I will have better News.
> 
> 
> 
> 

Thx, Archer

theArcher73, 2020-04-07:
> You may try BME680_I2C_ADDR_SECONDARY depending on you hardware (SDIO wiring)

Unfortunately no success :-(

cyberman54, 2020-04-07:
> I][i2c.cpp:23] i2c_scan(): Starting I2C bus scan...
> [I][i2c.cpp:71] i2c_scan(): I2C scan done, 0 devices found.

If the ESP32 i2c driver does not show the address of you BME680 during scan, data from BME sensor can't be read, thus the sensor is reported as not found.

You may have a problem with the i2c bus on your device / in your setup.

Take a closer look to the i2c bus: wiring, pullup resistors. Check bus signals with digital scope.
Is the display working? (is also using i2c)

theArcher73, 2020-04-08:
Hello,

since I currently can't get any further @twentysixer has kindly agreed to help me with troubleshooting. The pieces are already in a package in a Hermes truck through the republic. We will contact you as soon as we can continue, first of all have a nice Easter!

Archer

theArcher73, 2020-04-12:
So, I close the issue, sorry for your time.  Classical bug layer 8, the proof is given that a BME280 cannot be addressed with the library of the BME680. 

It's just crap when you're stewing in your own juice, as soon as someone looks at the chunks, such a stupid error is immediately obvious.

I wish you all a happy Easter,

Archer

closed 2020-04-12 by theArcher73