cyberman54/ESP32-Paxcounter #590: Downlink remote control TTN explanation

robomotic, 2020-04-13:
Hello!
I have a very simple operation to perform: 

I want to send a Remote Control operation on Port 2 to get the device status which is 0x81 and another command which is 0x16 to enable the battery level reading.

Would it be something along the lines of:

> 0x81 0x01
> 0x16 0x01

Then encode in BASE64?

How do I generate the payload bytes for the TTN console?

Cheers.

cyberman54, 2020-04-13:
you can just enter the command payload in plain text (hex) in the TTN console in downlink field. You can combine both commands in one downlink packet: 81 01 16 01

closed 2020-04-13 by cyberman54

ScholliYT, 2020-06-09:
Maybe add this to the Readme file? Wasn't clear to me at first.

cyberman54, 2020-06-10:
@ScholliYT You're welcome to post a PR for readme.md