cyberman54/ESP32-Paxcounter #595: fluctuating battery reading from TTGO board with USB power

robomotic, 2020-04-15:
Hello there again!

Board:
LILYGO® TTGO LORA V1.3 868Mhz ESP32 Chip SX1276 Module 0.96 Inch OLED Screen WIFI And Bluetooth Development Board

Setup:

OS: Darwin x64 18.7.0
VSS Code: Version: 1.43.2
Platform IO: Core 4.3.1 Home 3.1.1
TTN Server: Enterprise
Gateway: Kona Mikro Gateway
Board powered by USB
Steps:

Clone Git Project from PlatformIO
Change platform.ini to: halfile = ttgov21new.h

Issue:
The battery readings are fluctuating:

> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4146mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4334mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4230mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4176mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4142mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4284mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4302mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4176mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4292mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4144mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4192mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4206mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4124mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4266mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4204mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4242mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4234mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4232mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4242mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4278mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4274mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4204mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4314mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4088mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4262mV
> [I][cyclic.cpp:73] doHousekeeping(): Battery: 4248mV
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> 
> 

From the code I would expect a 0xFFFF:

> // line 4: Battery + GPS status + Wifi channel
> #if (defined BAT_MEASURE_ADC || defined HAS_PMU)
>     if (batt_voltage == 0xffff)
>       dp_printf(0, 4, FONT_SMALL, 0, "%s", "USB    ");
>     else if (batt_voltage == 0)
>       dp_printf(0, 4, FONT_SMALL, 0, "%s", "No batt");
>     else
>       dp_printf(0, 4, FONT_SMALL, 0, "B:%.2fV", batt_voltage / 1000.0);
> #33 

When powered by USB cable.

Are there some settings I have to change?

cyberman54, 2020-04-15:
Please update to current HEAD.
Battery code has been deeply reworked the last days.
Fluctuating readings in a range of ~200mV are normal, depending on peak power consumption of your device / settings.

robomotic, 2020-04-16:
Yes thanks,
just tested:

> [I][cyclic.cpp:70] doHousekeeping(): Battery: 99%
> [I][cyclic.cpp:70] doHousekeeping(): Battery: 97%
> [I][senddata.cpp:110] sendData(): Counter cleared
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA
> [I][lorawan.cpp:513] myRxCallback(): Received 1 byte(s) of payload on port 2
> [I][rcommand.cpp:313] get_batt(): Remote command: get battery voltage
> [I][lorawan.cpp:265] lora_send(): 2 byte(s) sent to LORA

On the uplink I read now 4110 is this still expressed as mV so this means 4.110 mv?

Cheers.

cyberman54, 2020-04-16:
correct, the battery status is sent to uplink either by MAC command DevStatusReq, then you get a value 0..255, or by application payload, then you get millivolts.

closed 2020-04-16 by cyberman54