cyberman54/ESP32-Paxcounter #596: DEVEUI identical in ABP mode for TTGO boards

robomotic, 2020-04-17:
Hello there,
I have now 10 TTGO boards configured in ABP mode.
I flashed the firmware and discovered that all of them have the same DEVEUI which of course is making a mess in the TTN uplink stream even though I have set different device id and device address of course.

I have assumed that every Lora chip should have a different MAC address which will generate a different DEVEUI but this does not seem the case.

How do I fix this? 
Do I have to look into some other libraries, maybe is not related to your firmware?

Cheers.

cyberman54, 2020-04-17:
Use OTAA.
Paxcounter generates unique DEVEUI from MAC address of ESP32 chip.

cyberman54, 2020-04-17:
PS: this is an issue tracker, not a user support forum. If you need support, you may ask in a relevant forum, e.g. TTN

closed 2020-04-17 by cyberman54

robomotic, 2020-04-17:
Apologies, I will use that forum from now on.
Out of curiosity should it not use the Semtech MAC address instead?
I guess I can figure out if is possible to get it via a registry read instead of the ESP32.
Cheers.

cyberman54, 2020-04-17:
There is no "MAC address" in the SX127x series chips by Semtech, neither any other kind of silicon id - at least no one which is documented an could be queried via SPI.