cyberman54/ESP32-Paxcounter #597: PaxCounter plus SDS011: sleeping-mode

AugustQu, 2020-04-23:
Hi,

I hope I found a solution for this problem (putting the SDS011-device into sleep-mode for 30 seconds, then waking up).

Currently there is a discussion in the forum for this library. I hope to convince the author to modify the code a bit.

I will inform you of the progress of this discussion.

cyberman54, 2020-04-24:
Could you post link to the discussion thread here? Thx.

AugustQu, 2020-04-24:
https://github.com/ricki-z/SDS011/issues/35

AugustQu, 2020-04-24:
Shall I post my modifications to the library here? So someone interested can do it without waiting.

cyberman54, 2020-04-25:
yes, why not

AugustQu, 2020-04-27:
one problem remained in the usage of the SDS011-sensor with the PaxCounter-Software on the TTGO-board: the sleep/wakup-mechanism did not work, at least it did not work here for me and others reported ths too.

I hope I found a solution but it needs some work (not much). Here are the 5 steps needed:

#1: download the latest version of this lib: https://github.com/ricki-z/SDS011

#2: in the directory of the PaxCounter-software create a directory in the lib-directory. I named it SDS011

#3: extract the data from step#1 into this directory. It should include 2 files: _SDS011.cpp_ and _SDS011.h_

#4: open the file _SDS011.cpp_ in an editor and enter these 2 modifications:
    in the beginning of the file you see an array with the name SLEEPCMD. Below this array enter:

```
        static const byte WORKCMD[19] = {
          0xAA, // head
          0xB4, // command id
          0x06, // data byte 1
          0x01, // data byte 2 (set mode)
          0x01, // data byte 3 (work)
          0x00, // data byte 4
          0x00, // data byte 5
          0x00, // data byte 6
          0x00, // data byte 7
          0x00, // data byte 8
          0x00, // data byte 9
          0x00, // data byte 10
          0x00, // data byte 11
          0x00, // data byte 12
          0x00, // data byte 13
          0xFF, // data byte 14 (device id byte 1)
          0xFF, // data byte 15 (device id byte 2)
          0x06, // checksum
          0xAB  // tail
        };
```
In the file _SDS011.cpp_ you will also find the implementation of the function _wakeup()_. Modify it into this form:
    
```
   void SDS011::wakeup() {
        //  sds_data->write(0x01);
        //  sds_data->flush();
          for (uint8_t i = 0; i < 19; i++) {
            sds_data->write(WORKCMD[i]);
          }
          sds_data->flush();
          while (sds_data->available() > 0) {
            sds_data->read();
          }
        }
```
Save the file to disk.

#5: in the PaxCounter-software you will find a file _platformio.ini_. In this file you will find an entry with the name _lib_dep_sensors_. Comment out one line by putting a semicolon in front of it. So it will look like this:
`    ; https://github.com/ricki-z/SDS011.git
`    
Save the file. Finished. Now you can recompile the whole software and test the behavior. 
    
In my tests it worked. The sensor was put to sleep for 30 seconds, then was started again with 30 seconds for warm-up. Then the values were read from the sensor and the sensor put to sleep-mode again.

For my tests I used these lines in the file _hal/ttgov21new.h_ :
```
    // SDS011 dust sensor settings
    #define HAS_SDS011 1      // use SDS011
    // used pins on the ESP-side:
    #define SDS_TX 15   // connect to RX on the SDS011
    #define SDS_RX 2    // connect to TX on the SDS011

```
Please try ths and report here if it works.

x-RoNiN-x, 2020-04-27:
[image]

My Paxcounter with SDS011 and BME680 is running since 07.Apr.2020 whit this code Sleepcmd and Workcmd

AugustQu, 2020-04-27:
During my test I used an Arduino Uno and an Arduino Mega 2560. With these boards the existing library (unmodified) worked like a charm.

But I did have problems with the TTGO-board for the PaxCounter-software.

In case you did not encounter these problems you do not have to apply the modifications I described. They are not needed.

x-RoNiN-x, 2020-04-27:
The data in the Picture ist made by TTGO Paxcounter whit modified code in the SDS011 Lib

x-RoNiN-x, 2020-04-27:
@AugustQu 

The only problem we have would be the runtime of the SDS! With a rest period of 30 seconds, the sensor lasts a maximum of 2 years, according to the manufacturer. expressed very carefully.

Have you already successfully sent data to ttn with the Paxcounter and the SDS?

Greets

AugustQu, 2020-04-28:
@x-RoNiN-x 
Hi,

yes, I have sent data to TTN. I tested it with the 3 of the formats possible for PAYLOAD_ENCODER. I stopped my work when I saw the data in hex in the console of TTN. I stopped here because I do not need the further processing of the data via TTN (I need the data on the SD-card).

I did this with and without the data coming from ths SDS011.

cyberman54, 2020-12-09:
A deep sleep mode is currently implemented, see https://github.com/cyberman54/ESP32-Paxcounter/issues/627

cyberman54, 2022-02-13:
@x-RoNiN-x can you make PR for this patch?

cyberman54, 2022-02-16:
solved in v3.2.0.

closed 2022-02-16 by cyberman54