cyberman54/ESP32-Paxcounter #598: How to debug TTN joining issues?

synth3, 2020-04-30:
Hi there,

first of all: Thanks for all the work - it is great to find such nice organized code! :)

I hope someone can help me with debugging the following issue:

- I have a TTGO T-Beam 0.7 Board (could be a fake one - I can't really tell)
- After performing the setup as documented, and flashing the firmware the board will never join the TTN

The only change I made to the `src/hal/ttgobeam.h` is that I commented out `#define HAS_DISPLAY 1` as this board has no display.

Here is the serial output I get from the board (the MAC stuff is removed with `[...]`):

```log
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.99, runmode 0
[I][configmanager.cpp:191] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:56] open_storage(): Opening NVS
[I][configmanager.cpp:61] open_storage(): Done
[I][configmanager.cpp:205] loadConfig(): NVRAM settings version = 1.9.99
[I][configmanager.cpp:223] loadConfig(): bsecstate = 0
[I][configmanager.cpp:228] loadConfig(): loradr = 5
[I][configmanager.cpp:236] loadConfig(): txpower = 14
[I][configmanager.cpp:244] loadConfig(): adrmode = 1
[I][configmanager.cpp:252] loadConfig(): screensaver = 0
[I][configmanager.cpp:260] loadConfig(): screenon = 1
[I][configmanager.cpp:268] loadConfig(): countermode = 0
[I][configmanager.cpp:276] loadConfig(): sendcycle = 30
[I][configmanager.cpp:284] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:292] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:300] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:308] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:316] loadConfig(): blescantime = 8
[I][configmanager.cpp:324] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:332] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:340] loadConfig(): rssilimit = 0
[I][configmanager.cpp:348] loadConfig(): payloadmask = 127
[I][configmanager.cpp:356] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:363] loadConfig(): Done
[I][i2c.cpp:23] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:71] i2c_scan(): I2C scan done, 0 devices found.
[I][main.cpp:209] setup(): PSRAM found and initialized
[I][main.cpp:243] setup(): Starting LED Controller...
[I][power.cpp:171] calibrate_voltage(): ADC characterization based on reference voltage stored in eFuse
[D][power.cpp:229] read_battlevel(): batt_voltage = 4238mV / batt_percent = 100%
[I][gpsread.cpp:36] gps_init(): Using serial GPS
[I][main.cpp:295] setup(): Starting GPS Feed...
[I][lorawan.cpp:301] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:304] lora_stack_init(): Starting LMIC...
[I][main.cpp:376] setup(): Starting Wifi...
[D][lorawan.cpp:475] myEventCallback(): JOINING         
[I][main.cpp:389] setup(): Starting Interrupt Handler...
[I][main.cpp:413] setup(): Starting Timers...
[I][main.cpp:471] setup(): Starting Timekeeper...
[I][timekeeper.cpp:150] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][timesync.cpp:50] timesync_request(): [1.810] Timeserver sync request started, seqNo#46
[I][main.cpp:478] setup(): Features: PSRAM LED BATT OTA GPS LORA FILTER PACKED WIFI BTN_PD
[D][lorawan.cpp:475] myEventCallback(): TXSTART         
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -90dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -91dBi -> salted MAC [...] Bytes left
[D][lorawan.cpp:475] myEventCallback(): RXSTART         
[D][lorawan.cpp:475] myEventCallback(): RXSTART         
[D][lorawan.cpp:475] myEventCallback(): JOIN_WAIT       
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -81dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -82dBi -> salted MAC [...] Bytes left
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2280 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2580 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 2616 bytes left | Taskstate = 2
[D][cyclic.cpp:49] doHousekeeping(): Gpsloop 1492 bytes left | Taskstate = 1
[D][cyclic.cpp:64] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2
[D][power.cpp:229] read_battlevel(): batt_voltage = 4186mV / batt_percent = 99%
[I][cyclic.cpp:70] doHousekeeping(): Battery: 99%
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -81dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -82dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -81dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -89dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -90dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -95dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -80dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -92dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -92dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -88dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -93dBi -> salted MAC [...] Bytes left
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2280 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2580 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 2616 bytes left | Taskstate = 2
[D][cyclic.cpp:49] doHousekeeping(): Gpsloop 1492 bytes left | Taskstate = 1
[D][cyclic.cpp:64] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[D][power.cpp:229] read_battlevel(): batt_voltage = 4188mV / batt_percent = 99%
[I][cyclic.cpp:70] doHousekeeping(): Battery: 99%
[I][senddata.cpp:110] sendData(): Counter cleared
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -92dBi -> salted MAC [...] Bytes left
[D][lorawan.cpp:475] myEventCallback(): TXSTART        1
[D][lorawan.cpp:475] myEventCallback(): RXSTART        1
[D][lorawan.cpp:475] myEventCallback(): RXSTART        1
[D][lorawan.cpp:475] myEventCallback(): JOIN_WAIT      1
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -34dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -33dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -84dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -82dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -95dBi -> salted MAC [...] Bytes left
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2280 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2580 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 2616 bytes left | Taskstate = 2
[D][cyclic.cpp:49] doHousekeeping(): Gpsloop 1492 bytes left | Taskstate = 1
[D][cyclic.cpp:64] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[D][power.cpp:229] read_battlevel(): batt_voltage = 4174mV / batt_percent = 99%
[I][cyclic.cpp:70] doHousekeeping(): Battery: 99%
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -76dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -76dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): new   WiFi RSSI -96dBi -> salted MAC [...] Bytes left
[V][macsniff.cpp:125] mac_add(): known WiFi RSSI -94dBi -> salted MAC [...] Bytes left
```

- What could be the cause?
- What can I do to find the cause (maybe add some additional debug output)?

cyberman54, 2020-05-01:
- Use `#define LMIC_DEBUG_LEVEL 2` in `lmic_config.h` to get debug messages from LMIC stack
- Check that you have TTN radio coverage at your test location
- Check that DIO1 line of LORA chip is connected on your board as set in hal file

synth3, 2020-05-02:
Thanks a lot for the hints!

- After letting the TTGO run for some time (20min I guess) I suddenly saw activation messages in the TTN console and the meta data suggested that the signal strength is close to unusable (`-120` to `-117`)
  - After another 10min I finally saw Incoming messages
- The additional output (from `LMIC_DEBUG_LEVEL 2`) did not show any errors

Network conditions:
- There is a gateway ~400m away in a densely populated urban area (Berlin/Wilmersdorf)
- Some messages in the console reached two gateways even though the other one "near" (according to the TTN map) is more than 1km away

My conclusion would be that the board may be usable but the antenna is bad (I used the antenna that came with the board and I have no gear for checking the antenna).

Can the issue be generalised to: "When there are no errors logged and the join does not succeed then it's most probably a problem with reception or a missing physical connection on the board"?

cyberman54, 2020-05-02:
If you saw incoming messages, you probably have a weak reception problem.
You can try login by ABP instead of OTA, since ABP does not need a downlink.

Try another antenna. Some china boards are sold with 433 MHz antennas.

synth3, 2020-05-05:
Thanks again for your help! :)

...It's a combination of questionable reception at my place and "not so good" tx/rx performance of the T-Beam board that I have (again - it might be a bad clone).

- In the meantime I got another LoRa-board and another antenna
  - The new board performs way more reliable in joining and getting messages thru to an gateway with the same antenna used (and according to the meta info in the TTN console it's 3dBm more - still far from good, but seems good enough)

Besides that, your suggestion of using ABP makes the T-Beam definitely more usable with a success rate of around 60%. :)

I will close this issue - maybe and hopefully it's useful for someone else too.

closed 2020-05-05 by synth3