cyberman54/ESP32-Paxcounter #604: Max charge current

fafu, 2020-05-09:
It seems that the max charge current on the TTGO v1.0 (with AXP192 PMIC) is limited to ~450mA. I know this is due to current limits in the AXP192 register set but I am basically interested what the system could handle or if there are further limitations for the max charge current.

cyberman54, 2020-05-09:
You're welcome to check features of the used AXP library and create a pull request to modify charging rates, e.g. by a #definition in hal file.

cyberman54, 2020-05-09:
I added charge current control in development branch, could ypu please test?
I'm not sure if it's working; you alread mentioned issue with AXP192 library
https://github.com/lewisxhe/AXP202X_Library/issues/18

cyberman54, 2020-05-09:
@fafu The APX202X_Library is buggy in the method which sets the charge current register.
I proposed a change:
https://github.com/lewisxhe/AXP202X_Library/pull/19

fafu, 2020-05-10:
@cyberman54 Thank you for providing a fix for the APX202X_Library which I tested on the TTGO T-Beam v1.0  successfully.

closed 2020-05-10 by fafu

cyberman54, 2020-05-11:
@fafu 

I am basically interested what the system could handle or if there are further limitations for the max charge current.

AXP192 has two current inputs: VBUS, rated 500(typ)/900(max)mA, and ACIN, rated 2000mA. On T-BEAM v10 only VBUS is used, and wired to USB SMT socket. Thus, we have hardware limitation to maximum power of 2,5 Watt.

I tested a T-BEAM with a USB power measurement device. If i set maximum charging current, the system draws 0,5A at 5,2V, so we see the 2,5 Watt here. So in this setup the high current charging modes of AXP192 will never be reached, since the chip has insufficient input power by USB.