cyberman54/ESP32-Paxcounter #607: ttgo lilygo  sx1276 starting lmic fails

llobiancoale, 2020-05-26:
Hello guys 
ive got this board without oled [https://www.amazon.com/SX1276-Bluetooth-Internet-Antenna-Development/dp/B07B6N56ZG](url)
and i cant get it working.
Ive tried to upload the whole ttgo - hal configs but i dont know why it wont work... 
First off setting HAS_Display to 0 wont let me build & upload due to errors (with 1 i can build and upload it but it crashes on boot since it cant find the oled-) only with has_display 2 i was able to build the ttgo hals and boot to the next failure :D 

I discovered that lmic is for the lora stack but thats it  XD

secco04, 2020-05-26:
when using the ttgov2 halfile i wont get the lmic failure but it seems not to send any data ...after some time i see lora queue is full at the serial monitor

secco04, 2020-05-26:
i managed to get rid of the lmic failure ... 
but it is still not sending any lora packets ... maybe it is using the wrong pins?

closed 2020-05-26 by secco04

cyberman54, 2020-05-26:
Probably the board has DIO1 line not connected. You must wire it to a GPIO as defined in the hal file.