cyberman54/ESP32-Paxcounter #612: Bluetooth: observations/questions

AugustQu, 2020-06-16:
Hi,

today I played with the Paxcounter enabling the sniffing of Bluetooth-devices.

Here are my observations/questions:

#1: enabling
in paxcounter.conf I set BLECOUNTER 1 and recompiled everything. In the display this text appears: BLTH:off
Looking into the file written on SD-card I always found 0 devices for bluetooth.
Finally I found these lines in the file configmanager.cpp:
`  if (nvs_get_i8(my_handle, "blescanmode", &flash8) == ESP_OK) {
      // AQ cfg.blescan = flash8;
      ESP_LOGI(TAG, "BLEscanmode = %d", flash8);
    } else {
      ESP_LOGI(TAG, "BLEscanmode set to default %d", cfg.blescan);
      saveConfig();
    }
`
Here the var. blescan will be set to 0 (was 1 before) and so bluetooth-scanning disabled. As you can see in the line above I made a comment out of this line et voila: now it seems to scan.

Why this? What is done here?

#2: the first test showed 2 devices. So I enabled bluetooth on my Smartphone and on my PC, but still 2 devices are found. Why this?
With Wifi the number of devices changed accordingly when I enable/disable Wifi on some devices.

PS: by looking through the code I found variables like macs_wifi, macs_ble and macs_total, but macs_total is unused (only set to 0).

cyberman54, 2020-06-16:
#1: BLE on/off status is persisted via non volatile RAM. Initially the mode is taken from BLECOUNTER setting in paxcounter.conf. If you reflash the device after flashing it with default BLE=off you need to alter the BLE mode, either by rcommand, or by clearing NV RAM. Clearing NV RAM during startup can be enforced by modifying the version number. Yes, this is a crappy logic. Pull requests for improvement are welcome.

#2 scan detection time and, thus, exploration rate, of Wifi and BLE sniffing differ. To compare exploration rates it is necessary to average samples over time.  Since ESP32 has only 1 RF radio, scanning of wifi devices has a 1:13 ratio, while BLE has 1:3, so BLE packets can be faster scanned, but it depends on the device, if/when a BLE packet is sent.

You could use the new corona warn app as a BLE packet generator.

cyberman54, 2020-06-16:
macs_total, indeed, is no longer used. I will delete it in development branch.

AugustQu, 2020-06-18:
I played a bit with the code. So I set WIFICOUNTER to 0 and compiled the code. The device now shows PAX: 0. Looking into the file I can see a lot of entries 0, but sometimes the boards sees 1.

On my PC I can see 3 devices (via hcitool scan).

Hmmmmm.

cyberman54, 2020-06-18:
Check the filters in blescan.cpp, Paxcounter tries to focus on handys. Your PC may see additional BLE devices like notebooks, audio devices, beacons, etc.

AugustQu, 2020-06-22:
OK, will do that but not before wednesday. 

Currently I'm testing with a smartphone with Corona-app (active) but the device says: 0 devices found.

cyberman54, 2020-06-22:
Did you deactivate the vendor filter? Maybe your smartphone has a MAC which is not in the vendors list, if so, the vendor filter ignores it.

Leepq, 2020-06-30:
Is that because of MAC Address Randomization?

cyberman54, 2020-07-01:
No, MAC address randomization does not occur for vendor OUI.

Leepq, 2020-07-02:
I meant the MAC address is Randomized usually since Andriod 8.0 and iOS 8. This randomization is real for WiFi. I think in the later versions, Bluetooth MAC Address got affected as well. Both WiFi and Bluetooth have the randomization currently.

Randomization affects the whole 6 bytes of the MAC Address; thus, it is harder to get the Vendor OUI of the device.

cyberman54, 2020-07-02:
@Leepq But for randomized the OUI local bit SHOULD be set, so you can detect this.
https://www.nctatechnicalpapers.com/Paper/2019/2019-mac-randomization-in-mobile-devices/download

Leepq, 2020-07-02:
Do you have the method to set this for detection?

cyberman54, 2020-07-02:
method is to test the bit :-)

mikethebee, 2020-09-12:
It would be helpful to mention "Clearing NV RAM during startup can be enforced by modifying the version number" in the README. This was the first time I had switched from WiFi to BT and wondered why BLTH:off was showing, ut hadn't realised the NVRAM issue until I found the answer here while looking at other issues.  Thanks for an interesting enhancement.

cyberman54, 2020-09-12:
We really need a better logic for saving settings to NVRAM. Suggestions or Pull Requests are welcome.

cyberman54, 2020-09-12:
BLE switching logic improved.
See https://github.com/cyberman54/ESP32-Paxcounter/commit/f2b6e12bc95692d1af515265e9e5fa325e9f89ba

closed 2020-09-12 by cyberman54