cyberman54/ESP32-Paxcounter #613: Der Bezeichner ""DAYLIGHT_TIME"" ist nicht definiert.

GeorgWolter, 2020-06-16:
V1.9.996 / TTGO Beam T22_V1.0, 20190612
Visual Studio Code / Version: 1.46.0

Folks,,
I am just running into this issue. And date / time are not getting updated on the display:

Der Bezeichner ""PAYLOAD_BUFFER_SIZE"" ist nicht definiert.
Der Bezeichner ""DAYLIGHT_TIME"" ist nicht definiert.
Der Bezeichner ""STANDARD_TIME"" ist nicht definiert.
Der Bezeichner ""HAS_LED"" ist nicht definiert.
Der Bezeichner ""SENDCYCLE"" ist nicht definiert.
Der Bezeichner ""HOMECYCLE"" ist nicht definiert.

Hopefully any Prrro here finds some time for sharing a hint :-)
Thank you!
[image]

cyberman54, 2020-06-16:
Did you install platform.io?

GeorgWolter, 2020-06-16:
Yes, platform.io (C4.3.4 / H3.2.3) works fine and VSC build / upload runs successfully too.
I am using same  platform.io and VSC for other projects as well. No trouble.
Interesting: There is no such error with the ESP32-Pax release from March.

cyberman54, 2020-06-16:
I can't reproduce this with platformio core 4.3.4.

GeorgWolter, 2020-06-16:
That's good news. Thx.
Do you have any idea I how could somehow solve the problem maybe?

AugustQu, 2020-06-18:
can't reproduce this.
platformio core: 4.3.4
Visual Studio Code 1.46.0

DAYLIGHT_TIME is used in main and defined in paxcounter.conf.

cyberman54, 2020-09-12:
Code was migrated to Platformio 5.0, this issue cannot be reproduced. Closing.

closed 2020-09-12 by cyberman54