cyberman54/ESP32-Paxcounter #614: Adding To Loriot Network

jamessugrue, 2020-06-18:
Hi 
Has anyone successfully added one of these devices to the Loriot network? 
In particular, I'm curious about how to use the parameters from Loriot in loraconf.h 

Any help would be appreciated! 
Thanks 
James

cyberman54, 2020-06-19:
It is possible, we had nodes in a Loriot Demo Installation running paxcounter.
By using OTAA (instead of ABP) join should be possible without tweaking parameters of LMIC stack.

But eventually Loriot network needs it's own network ID in the client's requests. If so, change in loraconf.h

`static const u1_t NETID = 0x13; // TTN`
to 
`static const u1_t NETID = 0x18; // Loriot*`

and rebuild and reflash the software.

*) according to this list
https://www.thethingsnetwork.org/docs/lorawan/prefix-assignments.html

jamessugrue, 2020-06-19:
Perfect - that worked. I moved the NETID constant out of the `#ifdef LORA_ABP` block.

closed 2020-06-19 by jamessugrue

cyberman54, 2020-06-19:
NETID is in the paxcounter code only used for ABP. So if you move it outside the #ifdef LORA_ABP block, this has no impact.

For OTAA & Loriot i think you just need to set register the DEVEUI of your device in the Loriot network, then you can join. No need to set NETID with OTAA.