cyberman54/ESP32-Paxcounter #617: Huge latency on Bw500Cr45Sf128 config

ddv2005, 2020-07-08:
Hello,

I can't get message rate more than 1 packet per 1-2 seconds even on Bw500Cr45Sf128 config. I have 2 devices (TBEAM v1.1) on COM6 & COM7. 
Send script on COM6 device:
```
import meshtastic
interface = meshtastic.StreamInterface("COM6") 
for x in range(6):
  interface.sendText(str(x),'^all', False, False) 
```
COM7 device monitored by meshtastic.exe --device COM7 --debug. See attached m.log.
I am expecting that COM7 device will receive all 6 messages during 1 seconds. But actually it take 1-2 seconds for every message.

[m.log](https://github.com/cyberman54/ESP32-Paxcounter/files/4887803/m.log)

cyberman54, 2020-07-08:
Does this issue relate to this repository??

ddv2005, 2020-07-08:
My bad. I am really sorry.

closed 2020-07-08 by ddv2005