cyberman54/ESP32-Paxcounter #618: Bluetooth zählt nicht

ChrSchultz, 2020-07-15:
mein TTGO Lora32 zeigt immer an BLTH:0 obwohl mein Handy bluetooth eingeschaltet ist. 
Log:
`[V][blecsan.cpp:115] gap_callback_handler(): BT payload rcvd -> type: 0x03 -> Complete List of 16-bit Service Class UUIDs
[V][blecsan.cpp:136] gap_callback_handler(): Device address (bda): 7b:58:26:80:53:50
[V][blecsan.cpp:138] gap_callback_handler(): Addr_type           : BLE_ADDR_TYPE_RANDOM
[V][blecsan.cpp:139] gap_callback_handler(): RSSI                : -90
[V][blecsan.cpp:152] gap_callback_handler(): BT device filtered

`

spmrider, 2020-07-15:
Einmal den MAC Address-Filter ausschalten bitte. Welcher Handyhersteller ist das?

cyberman54, 2020-07-15:
is local OUI bit set? Then it's a random local OUI.

ChrSchultz, 2020-07-15:
SAMSUNG J6
Ohne MAC-Filter zählt er.

cyberman54, 2020-09-23:
Thus, it's a vendor filter effect, the OUI will be missing in vendor_array.h

closed 2020-09-23 by cyberman54