cyberman54/ESP32-Paxcounter #619: SDS011 usage with TTGO T-Beam v1.0

B-Clever, 2020-07-20:
SDS011 support is a great feature, but I can´t get it to work with the T-Beam v1.0 (with AXP192). What steps are exactly needed to have the SDS011 Data visible in the TTN Console.

In my **ttgobeam10.h** file, I added: 

`#define HAS_SDS011 1 // use SDS011`
`// used pins on the ESP-side:`
`#define SDS_TX 1     // connect to RX on the SDS011`
`#define SDS_RX 3     // connect to TX on the SDS011`

The pins GPIO 1 and 3 seems to be the second hardware serial connection on the T-Beam. The first serial is for the GPS chip (Pins 34 and 12).
The ESP32 seems to have 3 hardware serial in total; one is for usb code upload. So the other two should be independently usable? When I connect the SDS011 to the Pins 1 and 3, the usb connection is not usable anymore. If I upload the code with SDS011 disconnected, open the serial monitor and connect the SDS011 afterwards, the serial monitor doesn´t output data anymore. The SDS011 fan is rotating for a while, then stops and is not starting a second time. 

Another question would be: How can I set the sleep time for the fan? For example to let it wakeup every 15 minutes to extend the lifetime.

As a side note: If the SDS011 is connected after code upload the payload send on slot 1 in TTN is not decodeable with the packed payload decoder code posted in the Read.me file. So I cannot have the serial monitor and SDS011 connected at the same time.

cyberman54, 2020-07-20:
Pins 1+3 are used for serial console, so you cannot use them at the same time for serial console.

B-Clever, 2020-07-22:
I tried it now with Pins 23 and 4. Now I am able to leave the sensor connected while still being able to upload code and see the serial monitor.
But the sensor is still not working. I tried these two configurations (Don´t know if the definition makes a difference):

`#define SDS_TX 23`
`#define SDS_RX 4 `

or

`#define SDS_TX GPIO_NUM_23`
`#define SDS_TX GPIO_NUM_4`

both didn´t work. 
When power is supplied, the fan starts spinning for a few seconds, then stops and don´t come up again. 

The serial console output with debug_level = 5

[image]

The SN of the SDS011 is 5003-3F3A in case this might be relevant. 

Think I read something about changed commands in newer versions.

B-Clever, 2020-07-24:
This issue seems to be related with [https://github.com/ricki-z/SDS011/issues/35](url). My SDS011 serial number contains numbers and letters, so it seems to be the new version with different commands.

closed 2022-02-16 by cyberman54