cyberman54/ESP32-Paxcounter #622: auf meinem Blog

AugustQu, 2020-08-03:
Hi,

ich habe da mal was dazu geschrieben:
https://linux-fuer-wi.blogspot.com/2020/08/menschen-zahlen.html

Schönen Gruß
AugustQ

cyberman54, 2020-08-03:
Danke! Gibt es Erkenntnisse zur "Genauigkeit"?

AugustQu, 2020-08-04:
Ich habe unterschiedliche Test durchgeführt, da mich das auch interessiert hat. Ich komme auf einen Faktor 2 - 3, genauer kann ich das nicht angeben.

Test: überschaubarer Platz, Anzahl Personen gezählt: 25. Paxcounter liefert zw. 6 und 10 als Zahl.
Test: Marktplatz in Wiesbaden. Anzahl Personen geschätzt: zwischen 50 und 100 (es waren mehr auf dem Platz, aber ich muß ja die Reichweite des ESP-WLANs berücksichtigen). Anzeige: 26 bis 38.

Berücksichtigen muß man, daß das in Wiesbaden ist. Wir sind hier nicht so Technik-affin.

closed 2020-09-05 by cyberman54