cyberman54/ESP32-Paxcounter #625: Time sync Node-RED server.

TheHysiK, 2020-08-20:
I am trying to sync time via NODE-RED timesync server but with M5core and Lopy4 I can't get it work.
I have configured NODE-RED as example described in readme and configured MQTT to point to AppID/devices/DeviceID/up 
I send a downlink to Port 2 with 87 as message and module receives it:
`][senddata.cpp:113] sendData(): Counter cleared
[D][lorawan.cpp:472] myEventCallback(): TXSTART
[I][lorawan.cpp:266] lora_send(): 4 byte(s) sent to LORA
[D][lorawan.cpp:472] myEventCallback(): RXSTART
[I][lorawan.cpp:510] myRxCallback(): Received 1 byte(s) of payload on port 2
[I][rcommand.cpp:332] set_time(): Timesync requested by timeserver
[D][lorawan.cpp:472] myEventCallback(): TXCOMPLETE`
but nothing is happening. I have tried with:
`#define TIME_SYNC_LORAWAN               0       // set to 1 to use LORA network as time source, 0 means off [default = 1]
#define TIME_SYNC_LORASERVER            1       /`
and with:
`#define TIME_SYNC_LORAWAN               1       // set to 1 to use LORA network as time source, 0 means off [default = 1]
#define TIME_SYNC_LORASERVER            0       /`

What I am doing wrong? Is there any bugs with NODE-RED time server?
Thanks in advance.

FlorianLudwig, 2020-08-20:
Hi @TheHysiK ,

do you see any messages on node-red? You can add debug nodes there to get messages printed inside a console.

TheHysiK, 2020-08-20:
Hi @FlorianLudwig thanks for your comment.
I have attached 3 images:
[https://imgur.com/a/VA8FNi6](url)
No debug message is shown at NODE-RED, if I add one at uplink I see message received by TTN with WIFI and BLE counters but no response of downlink message.

FlorianLudwig, 2020-08-20:
Hey @TheHysiK 
I just noticed you used
`TIME_SYNC_LORAWAN`
Which uses the lorawan mac commands to get the time. This does not interact with the node-red server at all.

`TIME_SYNC_LORASERVER` is for the node-red server. The only reason to use this is if you need higher (sub second) precision.

TheHysiK, 2020-08-20:
@FlorianLudwig I have tested with both of them; in the images I was using `TIME_SYNC_LORASERVER   1` so I guess that could not be the problem at all.

cyberman54, 2020-08-23:
@TheHsyik your screenshot show 3 different gateways, but only 1 of 3, the TTIG, provides absolute time. For syncing time you need to make sure, that the time sync request of the node is delivered by the TTIG, not by the other gateways.

TheHysiK, 2020-08-23:
> @TheHsyik your screenshot show 3 different gateways, but only 1 of 3, the TTIG, provides absolute time. For syncing time you need to make sure, that the time sync request of the node is delivered by the TTIG, not by the other gateways.

Hi @cyberman54 I understand that, I did all the test with that gateway and no other in range of the node, nothing happened.
Regards.

cyberman54, 2020-08-23:
please write serial log with debug level 4 and post it here.

TheHysiK, 2020-08-24:
> please write serial log with debug level 4 and post it here.

I checked it and it already was with Debug level 4, this is the code when Timesync is received:
`][senddata.cpp:113] sendData(): Counter cleared [D][lorawan.cpp:472] myEventCallback(): TXSTART [I][lorawan.cpp:266] lora_send(): 4 byte(s) sent to LORA [D][lorawan.cpp:472] myEventCallback(): RXSTART [I][lorawan.cpp:510] myRxCallback(): Received 1 byte(s) of payload on port 2 [I][rcommand.cpp:332] set_time(): Timesync requested by timeserver [D][lorawan.cpp:472] myEventCallback(): TXCOMPLETE`

cyberman54, 2020-08-24:
Log shows, timeserver answer is only 1 byte. This is probably 0xFF meaning that timeserver has no time to answer. Please check the payload on timesync port in TTN console and post here.

cyberman54, 2020-08-24:
Please check payload on all ports in TTN app console. Does the node send a timesync packet on port 9?

TheHysiK, 2020-08-24:
> Please check payload on all ports in TTN app console. Does the node send a timesync packet on port 9?

No, just a normal PAX counter message.

cyberman54, 2020-08-24:
> > Please check payload on all ports in TTN app console. Does the node send a timesync packet on port 9?
> 
> No, just a normal PAX counter message.

Did you wait some time? The action is triggered asynchronally, can take up to 30 seconds (depending on settings in paxcounter.conf)

cyberman54, 2020-09-02:
For some reason there was a ")" missing in the timserver routine. Not sure if this caused this issue.
I [commited](https://github.com/cyberman54/ESP32-Paxcounter/commit/18d15ba70931aa17582a26d286fd0bff99149aa8) a bugfix.
Please try again with current version.

I tested with my NODE RED timeserver, it worked.

TheHysiK, 2020-09-03:
> For some reason there was a ")" missing in the timserver routine. Not sure if this caused this issue.
> I [...]

@cyberman54 that was the problem, it is solved now. Thanks for your help.

closed 2020-09-03 by TheHysiK

cyberman54, 2020-09-03:
@theHysik sorry that had trouble with this bug. As your sync is working now, what precision do you achieve?

TheHysiK, 2020-09-07:
@cyberman54 I was checking today to give you a response and this is happening now:
`][senddata.cpp:113] sendData(): Counter cleared
[D][lorawan.cpp:472] myEventCallback(): TXSTART
[I][lorawan.cpp:266] lora_send(): 4 byte(s) sent to LORA
[D][lorawan.cpp:472] myEventCallback(): RXSTART
[I][lorawan.cpp:510] myRxCallback(): Received 1 byte(s) of payload on port 2
[I][rcommand.cpp:332] set_time(): Timesync requested by timeserver
[D][timesync.cpp:166] timesync_store(): [2163.286] seq#57[0]: t1=2163286
[W][timesync.cpp:215] timesync_serverAnswer(): [2163.293] Timeserver error: spurious data received
[D][lorawan.cpp:472] myEventCallback(): TXCOMPLETE     1
[D][timesync.cpp:166] timesync_store(): [2163.306] seq#57[0]: t0=2163306
[I][lorawan.cpp:266] lora_send(): 1 byte(s) sent to LORA
[D][lorawan.cpp:472] myEventCallback(): TXSTART
[D][lorawan.cpp:472] myEventCallback(): RXSTART
[I][lorawan.cpp:510] myRxCallback(): Received 6 byte(s) of payload on port 9
[D][timesync.cpp:166] timesync_store(): [2168.422] seq#57[0]: t1=2168422
[D][timesync.cpp:166] timesync_store(): [2168.424] seq#57[0]: t2=1599490788
[D][timesync.cpp:166] timesync_store(): [2168.430] seq#57[0]: t3=296
[W][timesync.cpp:104] timesync_processReq(): [2168.436] Timesync aborted: handshake out of sync
[D][lorawan.cpp:472] myEventCallback(): TXCOMPLETE
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2292 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2076 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:68] doHousekeeping(): LEDloop 480 bytes left | Taskstate = 1
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2292 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2076 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:68] doHousekeeping(): LEDloop 480 bytes left | Taskstate = 2
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 2292 bytes left | Taskstate = 0
[D][cyclic.cpp:42] doHousekeeping(): LMiCtask 2076 bytes left | Taskstate = 2
[D][cyclic.cpp:45] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:68] doHousekeeping(): LEDloop 480 bytes left | Taskstate = 2`
Thanks in advance