cyberman54/ESP32-Paxcounter #627: Which settings generate the lowest power consumption..

StefanBoehnert, 2020-08-25:
Hello,

I run the paxcounters on batteries for outdoor use and I want to know which settings provide low power consumption. I only need to read and send the wifi count every 5-10 minutes. Any recommondations?

cyberman54, 2020-12-09:
@StefanBoehnert good news: a deep sleep mode feature, setting the device to a power saving sleep mode between to send cycles, is currently implemented. You can already test the working alpha version by using the current code in development branch.

cyberman54, 2020-12-09:
With a T-BEAM v1.1 (with LORA chip HPD13 / SX1276), AXP192 in shutdown mode, LORA and GPS power cut off by AXP192, and a SSD1306 OLED connected to I2C pins and in sleep mode (because cutting off power of OLED would block i2c bus) i see

1,2 mA battery current, if esp wakeup is set by GPIO
0,9 mA battery current, if esp wakeup is set by timer
40-50 µA battery current, if esp is powered off by AXP192

closed 2020-12-21 by cyberman54