cyberman54/ESP32-Paxcounter #632: #ifdef HAL_ESP32_HAL_H_

AugustQu, 2020-09-05:
with the latest update of PlatformIO I was no longer able to compile the code. The problem was the statement 
`#ifdef HAL_ESP32_HAL_H_`
in the file OneBitDisplay.cpp.

This is te error-message:

> Compiling .pio/build/usb/lib0c7/OneBitDisplay/OneBitDisplay.cpp.o
.pio/libdeps/usb/OneBitDisplay/src/OneBitDisplay.cpp:51:23: fatal error: BLEDevice.h: No such file or directory

I had to undefine it explicitly. Then it compiles and the code worked.

cyberman54, 2020-09-05:
I rolled back OneBitDisplay to v1.5.0, this should work.
Besides this issue i discovered a pixel display bug in current OneBitDisplay v1.7.1, so it's better to avoid this version.

pbriesch, 2020-09-05:
With v1.5.0 it does not work either for me.

`.pio/libdeps/usb/OneBitDisplay@1.5.0/src/OneBitDisplay.cpp:47:23: fatal error: BLEDevice.h: No such file or directory`

pbriesch, 2020-09-05:
@AugustQu 
> I had to undefine it explicitly. Then it compiles and the code worked.

Sorry for my beginner question: How and in which file did you do this?

cyberman54, 2020-09-05:
Please post platformio version and platform (Win/Linux).

pbriesch, 2020-09-05:
Core 5.0.0·Home 3.3.0 on macOS

cyberman54, 2020-09-05:
allright, i never tested pio 5.0.0, still sticking to  4.3.4

vrees, 2020-09-07:
I can confim to this issue using following config:
PlatformIO: Core 5.0.0, Home 3.3.0
Linux, T-Beam 
ESP-IDF v4.3-dev-907-g6c17e3a64-dirty

As workaround I tried following statement in OneBitDisplay.cpp 
`#undef HAL_ESP32_HAL_H_`
Which helps to get rid of compiler error but leads to crashes after deploying to T-Beam.

cyberman54, 2020-09-07:
I tried to load ESP32 BLE library by adding a lib_deps statement in platformio.ini. Still compiler error with OneBitDisplay. Checked, that the files (BLEDevice.h, etc.) are in the there in the .pio build environment, and yes, there are.

Weird. Can anyone help?

nordominus, 2020-09-07:
I can also confirm on: 
PlatformIO: Core 5.0.0 Home 3.3.0
Windows, T-Beam v10

nordominus, 2020-09-07:
I got the code now compiles by hacking the "library.properties" file of OneBitDisplay in the folder ".pio\libdeps\usb\OneBitDisplay" and commenting the last line defining the "depends". And the firmware also works on my T-Beam (not yet a display attached)

_edited "library.properties" file of OneBitDisplay_
```
name=OneBitDisplay
version=1.5.0
author=Larry Bank
maintainer=Larry Bank
sentence=OLED and LCD library for 1-bit per pixel displays.
paragraph=Designed to provide a rich set of features with a simple API. Can automatically detect the display address (3C or 3D) and the controller type (SSD1306, SH1106 or SH1107). Includes 4 sizes of fixed fonts (6x8, 8x8, 16x16, 16x32). Can use I2C, SPI or bit bang I2C+SPI on any GPIO pins with the use of my BitBang_I2C library. Supports display sizes of: 128x128, 128x64, 128x32, 64x32, 96x16, 72x40. Includes scrolling text/gfx, deferred rendering, lines, sprites, circles, rectangles and a simple menu system.
category=Display
url=https://github.com/bitbank2/OneBitDisplay
architectures=*
includes=OneBitDisplay.h
;depends=BitBang_I2C
```

Maybe this information helps somehow?

AugustQu, 2020-09-07:
@pbriesch 
the compiler complained about an include in the file OneBitDisplay.cpp. So I added some lines:

#include <SPI.h>

#ifdef HAL_ESP32_HAL_H_            // added
#undef HAL_ESP32_HAL_H_          // added
#endif                                                // added

#ifdef HAL_ESP32_HAL_H_

The lines I added are between the include and the ifdef. Simply said: I undefd this define. Then it compiled and the code worked for me.

My environment:
Visual Studio 1.48.2
PlatformIO: 5.0.0 Home 3.3.0
Ubuntu 18.04

pbriesch, 2020-09-07:
@AugustQu

Thanks, is working now!

nordominus, 2020-09-07:
> 
> 
> I got the code now compiling by modifing the file `<project-roo>\.pio\libdeps\usb\OneBitDisplay\library.properties` (OneBitDisplay library) by commenting or deleting the line `depends=BitBang_I2C`:
> 
> 
> ```
> name=OneBitDisplay
> version=1.5.0
> author=Larry Bank
> maintainer=Larry Bank
> sentence=OLED and LCD library for 1-bit per pixel displays.
> paragraph=Designed to provide a rich set of features with a simple API. Can automatically detect the display address (3C or 3D) and the controller type (SSD1306, SH1106 or SH1107). Includes 4 sizes of fixed fonts (6x8, 8x8, 16x16, 16x32). Can use I2C, SPI or bit bang I2C+SPI on any GPIO pins with the use of my BitBang_I2C library. Supports display sizes of: 128x128, 128x64, 128x32, 64x32, 96x16, 72x40. Includes scrolling text/gfx, deferred rendering, lines, sprites, circles, rectangles and a simple menu system.
> category=Display
> url=https://github.com/bitbank2/OneBitDisplay
> architectures=*
> includes=OneBitDisplay.h
> ;depends=BitBang_I2C
> ```
> 
> Maybe this information helps somehow?

As described in my workaround there is a difference when building:

Without the depends in library.properties platformio recognizes that OneBitDisplay lib depends on "ESP32 BLE Arduino"
`|-- <OneBitDisplay> 1.5.0 (E:\coding\ESP32-Paxcounter\.pio\libdeps\usb\OneBitDisplay)
|   |-- <BitBang_I2C> 2.1.1 (E:\coding\ESP32-Paxcounter\.pio\libdeps\usb\BitBang_I2C@2.1.1)
|   |   |-- <Wire> 1.0.1 (C:\Users\Andreas\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
|   |-- <ESP32 BLE Arduino> 1.0.1 (C:\Users\Andreas\.platformio\packages\framework-arduinoespressif32\libraries\BLE)`

Without that change platformio fails to recognize the dependecy and does not add "ESP32 BLE Arduino"
`|-- <OneBitDisplay> 1.5.0 (E:\coding\ESP32-Paxcounter\.pio\libdeps\usb\OneBitDisplay)
|   |-- <BitBang_I2C> 2.1.1 (E:\coding\ESP32-Paxcounter\.pio\libdeps\usb\BitBang_I2C@2.1.1)
|   |   |-- <Wire> 1.0.1 (C:\Users\Andreas\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
|   |-- <SPI> 1.0 (C:\Users\Andreas\.platformio\packages\framework-arduinoespressif32\libraries\SPI)`

nordominus, 2020-09-07:
> I got the code now compiling by modifing the file `<project-roo>\.pio\libdeps\usb\OneBitDisplay\library.properties` (OneBitDisplay library) by commenting or deleting the line [...]

As described in my workaround there is a difference when building:

Without the depends in library.properties platformio recognizes that OneBitDisplay lib depends on "ESP32 BLE Arduino"
> |-- <OneBitDisplay> 1.5.0 (E:\ESP32-Paxcounter\.pio\libdeps\usb\OneBitDisplay)
>       |   |-- <BitBang_I2C> 2.1.1 (ESP32-Paxcounter\.pio\libdeps\usb\BitBang_I2C@2.1.1)
>       |   |   |-- <Wire> 1.0.1 (C:\Users\user\\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
>       **|   |-- <ESP32 BLE Arduino> 1.0.1 (C:\Users\user\\.platformio\packages\framework-arduinoespressif32\libraries\BLE)**
>       |-- <SPI> 1.0 (C:\Users\user\\.platformio\packages\framework-arduinoespressif32\libraries\SPI)

Without that change platformio fails to recognize the dependecy and does not add "ESP32 BLE Arduino"
>  |-- <OneBitDisplay> 1.5.0 (E:\ESP32-Paxcounter\.pio\libdeps\usb\OneBitDisplay)
>        |   |-- <BitBang_I2C> 2.1.1 (E:\ESP32-Paxcounter\.pio\libdeps\usb\BitBang_I2C@2.1.1)
>        |   |   |-- <Wire> 1.0.1 (C:\Users\user\\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
>        |   |-- <SPI> 1.0 (C:\Users\user\\.platformio\packages\framework-arduinoespressif32\libraries\SPI)

nordominus, 2020-09-07:
Okay, somehow the [Library Dependency Finder (LDF)](https://docs.platformio.org/en/latest/librarymanager/ldf.html) of Platform.io misses to detect the dependencies of OneBitDisplay.

Add "lib_ldf_mode = deep" to your platformio.ini in the [env] section, like this

[image]

Clean (or delete whole .pio folder), build again and it should compile now

cyberman54, 2020-09-08:
Thanks for investigation and fixing! Tested & Merged.

PR #634 (2020-09-08): Fix for #632 - LDF will recognize BLE dependency of OneBitDisplay correctly
Fixes #632
closed 2020-09-08 by cyberman54