cyberman54/ESP32-Paxcounter #633: Rebooting

AugustQu, 2020-09-05:
during my test today I got a lot of events like these:

> [2020-09-05 12:09:28] [D][macsniff.cpp:132] mac_add(): new   BLTH RSSI -94dBi -> salted MAC 77C15EA7 -> E (15138) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[2020-09-05 12:09:28] E (15138) task_wdt:  - IDLE0 (CPU 0)
[2020-09-05 12:09:28] E (15138) task_wdt: Tasks currently running:
[2020-09-05 12:09:28] E (15138) task_wdt: CPU 0: Btc_task
[2020-09-05 12:09:28] E (15138) task_wdt: CPU 1: IDLE1
[2020-09-05 12:09:28] E (15138) task_wdt: Aborting.
[2020-09-05 12:09:28] abort() was called at PC 0x400e932b on core 0
[2020-09-05 12:09:28] 
[2020-09-05 12:09:28] Backtrace: 0x4009374c:0x3ffbe320 0x4009397d:0x3ffbe340 0x400e932b:0x3ffbe360 0x400858dd:0x3ffbe380 0x40009217:0x3ffd63b0 0x40007d13:0x3ffd63d0 0x40007c69:0x3ffd63f0 0x40008106:0x3ffd6410 0x400e764b:0x3ffd64a0 0x400d5212:0x3ffd6500 0x400811b9:0x3ffd6570 0x40168f59:0x3ffd65a0 0x40164046:0x3ffd65c0 0x4008f82d:0x3ffd65f0
[2020-09-05 12:09:28] 
[2020-09-05 12:09:28] Rebooting...
[2020-09-05 12:09:28] ets Jun  8 2016 00:22:57
[2020-09-05 12:09:28] 
[2020-09-05 12:09:28] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[2020-09-05 12:09:28] configsip: 188777542, SPIWP:0xee
[2020-09-05 12:09:28] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2020-09-05 12:09:28] mode:DIO, clock div:2
[2020-09-05 12:09:28] load:0x3fff0018,len:4
[2020-09-05 12:09:28] load:0x3fff001c,len:1044
[2020-09-05 12:09:28] load:0x40078000,len:8896
[2020-09-05 12:09:28] load:0x40080400,len:5828
[2020-09-05 12:09:28] entry 0x400806ac

What does this mean?

cyberman54, 2020-09-05:
Probably some part of the program consumes too much CPU time. Then the ESP32 wachtdog ist not feeded in time, thus it resets the CPU. If this happenes in a dense RF environment with many Wifi/BLE devices sending data packets, it's probably caused by ineffective code which consumes too much CPU cycles while analyzing the received data.

AugustQu, 2020-09-09:
I made another test today. Over 7 minutes I encountered 45 reboots..... Sometimes I worked for 10 seconds. In this time it counted approx. 60 to 80 BLE-devices (Wifi is disabled).

cyberman54, 2020-09-09:
Reboots are sometimes triggered by a power problem. Wifi/BLE sniffing is causing high peak consumptions. Can you test to change your power supply?

I did a BLE scan test yesterday in a dense environment with current code and T-BEAM. It captured ~5000 beacons in a few minutes, no crashes.

AugustQu, 2020-09-10:
I made the test with the ttgov21new-file (=the corresponding board). The board was connected to my Notebook for capturing the messages.

And in these 7 minutes I captured >40.000 lines of info.

Can this cause the problem?

Where in the code is the watchdog-timer set? Can I simply set a higher value before rebooting?

cyberman54, 2020-09-10:
@AugustQu if you use log level=DEBUG while routing the log messages over UART (USB) port, this sure can and will trigger the watchdog when heavy traffic arrives. The watchdog is a [feature of ESP32 IDF platform](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html). The esp32 arduino framework has an [API ](https://github.com/espressif/arduino-esp32/pull/2248/commits/b8f8502fcb02bd845ce11de26037827507a25232)to modify the ESP32 IDF watchdog settings during runtime.

cyberman54, 2020-09-12:
@AugustQu please try again with current code, it was upgraded to ESP IDF 4.1

AugustQu, 2020-09-25:
@cyberman54 
Please close this topic.

The cause is the amount of data I'm writing to the serial interface. Reduce these logging-lines and the reboots are gone.

closed 2020-09-25 by cyberman54