cyberman54/ESP32-Paxcounter #635: Guru Meditation Crash with Platformio-Espressif V2.0.0

cyberman54, 2020-09-08:
Updating platformio esp32 platform to [v2.0.0](https://github.com/platformio/platform-espressif32/tree/v2.0.0) makes the code crashing with Guru Meditation error after ~2 minutes runtime.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d55a1  PS      : 0x00060a30  A0      : 0x800d56bc  A1      : 0x3ffd6120
A2      : 0x3ffc6dec  A3      : 0x3ffd6156  A4      : 0x3f80afc4  A5      : 0x00000001
A6      : 0x0000cd5e  A7      : 0x3ffc6df0  A8      : 0xad567800  A9      : 0x00000001
A10     : 0x01ca1000  A11     : 0x3ffd6169  A12     : 0x00000080  A13     : 0x3f40372b
A14     : 0x00000007  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xad567810  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400d55a1:0x3ffd6120 0x400d56b9:0x3ffd6150 0x400813ee:0x3ffd6190 0x4016c799:0x3ffd61b0 0x40167886:0x3ffd61d0 0x40091de1:0x3ffd6200

Rebooting...
```

nordominus, 2020-09-10:
Did you try to activate the esp32_exception_decoder via the [monitor_filters](https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#filters) variable in platformio.ini to get a decoded stacktrace?
> monitor_filters = time, esp32_exception_decoder, default

And which steps did you do to update to v2.0.0?

cyberman54, 2020-09-11:
Thanks for the hints. Meanwhile the code seems to work stable with v2.0.0.
I will push an update for platformio.ini soon.

closed 2020-09-11 by cyberman54