cyberman54/ESP32-Paxcounter #649: Paxcounter sending data every 60 sec but sendcycle is set to 120

ChrSchultz, 2020-09-29:
My Paxcounters sendcycle is set to 120 

    {
      "blescantime": 8,
      "flags": {
       "adr": 1,
       "alarm": 0,
       "antenna": 0,
       "blescan": 0,
       "countermode": 0,
       "filter": 0,
       "screen": 1,
       "screensaver": 0
    },
       "loradr": 5,
       "payloadmask": {
       "alarm": 1,
       "battery": 1,
    "bme": 1,
    "counter": 1,
    "gps": 1,
    "sensor1": 1,
    "sensor2": 1,
    "sensor3": 1
    },
    "rgblum": 30,
    "rssilimit": 0,
    "sendcycle": 120,
    "txpower": 14,
    "version": "2.0.12",
    "wifichancycle": 50
    }
but instead of sending every 240 seconds he is sending data every 60 seconds...

cyberman54, 2020-09-30:
Default is 60 seconds, as set in [paxcounter_orig.conf](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/paxcounter_orig.conf#L13).

If you change the cycle by rcommand, this should be persistent, meaning after reset device keeps this setting.

Need to check, if i can reproduce your observation, and if yes, will try to spot the bug.

cyberman54, 2020-09-30:
I can confirm this issue, i could reproduce it.
But could not spot the bug yet.

Currently, it seems it's NOT a bug in the NVS saving routine, since a modified runtime setting for counter mode is succesful restored after power up.

Weird.

cyberman54, 2020-10-01:
I just pushed a fix, which probably solves this, to development branch (v2.0.16).
Could you, please, retest with this version and report result here?
If the fix is successful, i will merge it to master.
Thanks.

/cc @spmrider

ChrSchultz, 2020-10-04:
I can´'t find development v2.0.16 ... in your repository...

cyberman54, 2020-10-04:
I finally spotted the bug, in main.cpp the sendcycle was always initalized with the default value, not with the runtime configuration value. This is now fixed in master and now it should work. Can you confirm, please?

cyberman54, 2020-10-04:
https://github.com/cyberman54/ESP32-Paxcounter/commit/114aff924675f09c717fadc8305eb578f3325de8

ChrSchultz, 2020-10-04:
now It works fine, I close Issue thanks

closed 2020-10-04 by ChrSchultz