cyberman54/ESP32-Paxcounter #652: Timesync fails

ChrSchultz, 2020-09-30:
I tried to sync time via timeserver but I the Node red timeserver "answers no confident time" and sync abortet with no handshake.
I called rccommand 0x87

=Nodered msg before encoding to base64
   payload: object
   payload_raw: buffer[1]raw
   0: 0xff
   port: 9

cyberman54, 2020-10-01:
This is what it says it is: Your timeserver does not see a gateway with a confident time. So won't do the timesync.
Check the timestamp of the gateway(s) which serve your node.

closed 2020-10-01 by cyberman54

ChrSchultz, 2020-10-01:
I fixed this with correction in Timeserver code: `server_time=new Date(); ` new Date(msg.payload.metadata.time) didn't work.
Node Red runs on my Raspi 3. and the timeserver takes the time from the Raspi. :-)

cyberman54, 2020-10-01:
i don't overlook this currently, can you make a PR for this and comment your changes?

cyberman54, 2020-10-02:
Using local timeserver time instead of lorawan server time to validate gateway time will work, but breaks the concept to get time only from the network. It means that your timeserver needs it's own timebase, thus not only working as a broker.

I don't see the root cause why this doesn't work within your lorawan network server.
What kind of lorawan network are you using?

Please check the lorawan packet on the gateway, whichs conveys the timesync payload for the node. What is displayed in the field "timestamp"? (This field shows the lorawan server time, which is used to check if a seen gateway's timestamp is recent.)

ChrSchultz, 2020-10-02:
I think that the differece between gateway_time and server_time was too great >1000ms.. so it reports bad confidence. was fixed if i restartet my gateway... :-) now it works with `server_time = new Date(msg.payload.metadata.time); ' 👍