cyberman54/ESP32-Paxcounter #656: heltec boot looping

dh47, 2020-10-05:
Following this guide and programming with the Heltec V1.

I get a screen showing the pax counter but then the screen disappears and goes blank then shows again boot looping?

After monitoring in serial I get this error.

10:00:47.357 > [I][lorawan.cpp:203] get_hard_deveui(): Could not read DEVEUI from serial EEPROM

cyberman54, 2020-10-06:
This looks like you compiled the software for board "generic.h", instead for "heltec.h".
Check your settings in platformio.ini. If you're new to this project, follow installation steps in readme.md carefully.
You MUST edit some files before compiling the software.

dh47, 2020-10-06:
Ok, thanks I'll try that.

closed 2020-10-06 by cyberman54

dh47, 2020-10-06:
Ok, I have followed the guide again and made sure I had the board selected in that ini file.  I get further this time with the main screen and then what looks to be a QR code, but then nothing after that.

dh47, 2020-10-06:
Master.

dh47, 2020-10-06:
Hangs on.
`10:18:25.506 > [I][lorawan.cpp:305] lora_stack_init(): Starting LMIC...`

cyberman54, 2020-10-07:
looks like a wrong pin mapping. Please specify exact version of heltec board you are using and post a high res picture of the board here.

dh47, 2020-10-07:
WiFi LoRa 32

[image]
[image]

cyberman54, 2020-10-07:
I tested current code in master with a Heltec v1 board which has same shape as the board shown here. No issues, software starts and joins.

Check your board version and matching settings in heltec.h carefully, e.g. radio chip type. Try CFG_SX1272_radio (instead of 1276) in heltec.h

cyberman54, 2020-10-07:
PS: your board has not the printing "868-915Mhz" on the backside. So it probably will have radio chip SX1272. You need to adjust heltec.h as explained above. Software should work then, but you won't be able to use 868 MHz LORAWAN networks with this version. Depends on you country and LORAWAN provider.

dh47, 2020-10-07:
Hi, it was sent to me from a friend who says its the 433MHz model, but he's not sure.  I will try and compile with the SX1272.