cyberman54/ESP32-Paxcounter #658: No BATTERY status for T-Beam on paxcounter 2.0.x

PlumbumDN, 2020-10-07:
After flashing paxcounter 2.0 on Sunday I experienced that my TTGO T-Beam has no battery status anymore.
You can see, that after flashing the cwa counter became active and battery voltage was missing.

I checked several things and measured the correct voltage on GPIO 35, thus I think the config and PIN mapping is correct.
Anyone else experiencing this?

[image]

cyberman54, 2020-10-07:
What T-Beam version are you using? pre 1.0, 1.0 or 1.1.?

PlumbumDN, 2020-10-07:
its a pre 1.0 (late 2018) T-Beam.

cyberman54, 2020-10-07:
could you please try to change line122 in power.h from &sigmoidal to &linear and re-test?

cyberman54, 2020-10-07:
And: which software version did you use before, that did show correct battery voltage values?

PlumbumDN, 2020-10-07:
ok, change it to linear, but no values at all. 
[image]
Previous version on T-Beam was quite old... 1.8.2 if build logs are correct... but could be the result of not updating platformio.ini in the past.
Because of the big gap I reconfigured everything from scratch in 2.0.1x, cleared esp and did a fresh build

cyberman54, 2020-10-07:
I will check this on a old T-Beam. Generally, battery monitoring was refactored and improved since 1.8.x, thus it is possible that a bug came in. The relevant code is in power.cpp

PlumbumDN, 2020-10-07:
No hurry, just noticed the missing values ;-)

cyberman54, 2020-10-09:
Turns out: it's not a bug, it's a feature.

With newer versions of paxcounter it is possible to control sending of device data more granular. See [description](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/README.md#Sensors%20and%20Peripherals).
The default for battery values is now OFF. You can it turn it ON by once sending rcommand 0x16 0x01 to the device. The setting is stored in NVRAM then and thus will be persistant, until you change it again.

closed 2020-10-09 by cyberman54