cyberman54/ESP32-Paxcounter #663: Clarify how to enable BME280 sensor

Noki, 2020-10-29:
Hi!

I'm trying to connect a BME280 to a TTGO Lora32 V2.1 NEW but I can't get it to work.

I connected the sensor and changed the section in generic.h to:

```
// BME680 sensor on I2C bus
// #define HAS_BME 1 // Enable BME sensors in general
// #define HAS_BME680 GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL
// #define BME680_ADDR BME680_I2C_ADDR_PRIMARY // connect SDIO of BME680 to GND

// BME280 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME280 GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL
#define BME280_ADDR 0x76 // change to 0x77 depending on your wiring
```

In the debug window I can see that the device was found and that 0x76 is the correct address:

```
[I][i2c.cpp:22] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:43] i2c_scan(): 0x3C: SSD1306 Display controller
[I][i2c.cpp:48] i2c_scan(): 0x76: Bosch BME MEMS
[I][i2c.cpp:74] i2c_scan(): I2C scan done, 2 devices found.
```

From what I can see in platformio.ini I don't think the following part mentioned in generic.h is required anymore, since lib_deps_sensors is part of lib_deps_all:

```
// in platformio.ini append
// lib_deps = <...> ${common.lib_deps_sensors}
// for loading necessary libraries
```

I can't see a screen with BME values and the values are also not transmitted to TTN. The normal paxcounter functionality works fine.

I tried adjusting the BME680 STATE_SAVE_PERIOD because I wasn't sure if data for the sensor get's only transmitted 4 times a day, but it seems to have no effect.

```
// Settings for BME680 environmental sensor
#define BME_TEMP_OFFSET                 5.0f    // Offset sensor on chip temp <-> ambient temp [default = 5°C]
#define STATE_SAVE_PERIOD               UINT32_C(360 * 60 * 1000) // update every 360 minutes = 4 times a day
#define BMECYCLE                        1       // bme sensor read cycle in seconds [default = 1 secs]
```

Can someone please point out the missing steps to get the sensor to read, display and transmit data and to adjust the update interval?

Best regards
Tobias

cyberman54, 2020-10-29:
`STATE_SAVE_PERIOD` is for BME680 only and controls how often a snapshot of calibration data is stored to NVRAM. You don't need this for BME280 and it has no impact on your issue.

What is prompted during startup:

`BME280 sensor found and initialized`
or
`BME280 sensor not found`
or 
`I2c bus busy - BME280 initialization error`

or none of the above messages?

cyberman54, 2020-10-30:
@gregorwolf Can you help here?

cyberman54, 2020-10-30:
This is an open source project, you're welcome to contribute to the documentation!

jsponz, 2020-12-21:
@Noki Could you share your file? I am having some issues with Heltec Lora ESP32 v2 and I2C. I am using pins 21 and 22, and it does not read any data. 

Thanks!

cyberman54, 2020-12-22:
@jsponz did you look at the pinout specs for your heltec v2? You can't use #21 as SDA, since this GPIO is used for Vext control. Use the i2c bus of display, instead (SDA_OLED / SCL_OLED in hal file).

jsponz, 2020-12-22:
Thanks! You are right. I have changed to pins 15 and 4 and it works :-)