cyberman54/ESP32-Paxcounter #667: Eco poer board - stops running when starting IRQ handler

dl5dla, 2020-10-31:
Hi,
I try to run the code on an ECO Power board. I enabled ecopower.h in the platformio file, and could compile the code successfully. But when running the code, the console output stops at the point where the IRQ handler is starting. Just one lora packet is sent out.

15:06:21.744 > I (509) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=1
15:06:21.749 > 
15:06:21.750 > W (515) wifi: ap start fail
15:06:21.752 > 
15:06:21.752 > I (517) wifi: Set ps type: 0
15:06:21.755 > 
15:06:21.804 > I (579) wifi: ic_enable_sniffer
15:06:21.807 > [I][main.cpp:415] setup(): Starting Wifi...
15:06:21.810 > I (581) wifi: mode : null
15:06:21.813 > [I][main.cpp:430] setup(): Starting In�

Pin settings of the lora SPI:
#define LORA_SCK  SCK                                                                                                                       
#define LORA_CS   SS        
#define LORA_MISO MISO
#define LORA_MOSI MOSI
#define LORA_RST  (17)
#define LORA_IRQ  (16)
#define LORA_IO1  (26) // manually wired
#define LORA_IO2  LMIC_UNUSED_PIN

I also tried to disable lora to make sure, that the pin settings are not affectiing the run, but got the same console output.

Any idea, what's going wrong here?

cyberman54, 2020-10-31:
I re-tested current code on a Eco-Power board and can reproduce this issue. It's probably a RAM shortage, maybe this came in while the arduinio-esp32 framework is growing over time.

I found out that this issue can be avoided by deactivating IF482 serial line feature in ecopower.h:

// Settings for IF482 interface
//#define HAS_IF482 9600, SERIAL_7E1, GPIO_NUM_3, GPIO_NUM_1 // RX, TX

Changed this in master.

Could you, please, modify your ecopower.h and re-test? Thanks.

dl5dla, 2020-11-01:
Thanks a lot. Yes, with these changes in ecopower.h it works.

But the device does not report the ens count. It just sends an empy lora packet on port 10 without payload. For comparison, I compiled the code for a TTGo T-beam (with no other changes than enabling ttgobeam10.h) and that device sends the ens values.

cyberman54, 2020-11-01:
Check your NVRAM runtime setting for payloadmask and your defaults in paxcounte4.conf.

Did you compile the code with ENS option defined?

cyberman54, 2020-11-01:
The missing ENS count should be fixed now, please look at issue #668 https://github.com/cyberman54/ESP32-Paxcounter/issues/668

dl5dla, 2020-11-01:
Thanks again :-) Meanwhile I had cleared the NVRAM (by incrementing the version number in platformio.ini). That already had fixed the problem. But anyway moved to the new version now

cyberman54, 2020-11-01:
So, back to the original issue. Any ideas how to debug this?

dl5dla, 2020-11-01:
You mean the memory shortage, when IF482 is activated? Sorry, currently no idea - I am not very familar with the ESP32, but will think about it.

cyberman54, 2020-11-02:
I think we can close this. The pitfall was that in ecopower.h default setting IF482 serial interface was enabled, using the serial console pins 3 (RX) and 1(TX). If you have USB cable connected using the external progamming board, there is conflict since the programming board is using UART on same pins. This is causing a crash of the IF482 serial output. Since this output is triggered by interrupt, the crash happenes with the first interrupt.

closed 2020-11-02 by cyberman54