cyberman54/ESP32-Paxcounter #669: Add Code and Option for SCD30

sbamueller, 2020-11-01:
The PaxCounter already allows for a variety of sensors to be attached. Eg. you can connect a BME680 and read such intersting things like Indoor air quality index. Wouln't it be great to have a SCD30 CO2 option included to measure CO2 Levels and thus be able to detect when its necessary to ventilate the room?

There are a variety of libraries available: https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library and https://github.com/paulvha/scd30.

Especially durin the Covid pandameic it is becomming much more intresting to add a refrence or a measurement for people in a room and air quality.

hansju-11, 2021-03-17:
Hello sbamueller,

I have a quick and dirty version based on paxcounter 2.4.0 as sensor 2. I just started to clean up it a little and extend it if 3 NEO pixel, so I can change the color depend on c02 level. 

Do you have some specific request?

cyberman54, 2021-03-18:
Could you add your solution as PR to development branch?

hansju-11, 2021-04-01:
I put a PR on development branche today  (#780) . So all of you can have a look on it.

hansju-11, 2021-04-08:
I put a PR on dev branch today. It is a more complex way (like sds011 sensor).