cyberman54/ESP32-Paxcounter #688: RGB LED always red

jsponz, 2020-12-21:
I have installed the last version in Heltec Lora ESP32 V2. 

I connect it to TTN. 

It works, but I have observed two things: 

- RGB LED is always red. According to documentation that means: "LoRaWAN stack error". Is that possible? BTW, what should be the default status? 
- Gateway is 10m away (test phase), but I don't receive all the messages. I see in the terminal "lora_send(): 4 byte(s) sent to LORA", but I don't get always the message in TTN. 

Are the two things releated? 

Thanks!

cyberman54, 2020-12-21:
Heltec v2 has no RGB LED!?

cyberman54, 2020-12-21:
Note: orange LED on Heltec V2 boards is for power, and is driven hy hardware, not by paxcounter.

If you have problems with LORAWAN / LMIC on a Heltec V2 board, please consult google, there are some entries in forums with possible issues.

Make sure you got board with matching frequency for TTN in your country (e.g. 868 MHz, not 433 MHz).
Make sure you are not using a single channel LORA gateway; those are not LORAWAN compliant and will cause packet loss.

closed 2020-12-21 by cyberman54

jsponz, 2020-12-21:
Thanks for the info.