cyberman54/ESP32-Paxcounter #712: Can not get GPS fix with LoPy4 and Pytrack or Pytrack 2.0 x

tjvandam, 2021-01-07:
I have been trying this tutorial: https://www.hackster.io/simon-kemper/how-to-build-a-lorawan-smartphone-counter-with-esp32-1c22b6 with brand new LoPy4 and old Pytrack and Pytrack 2.0 x

I have uncommented the GPS lines in the hal/lopy4.h file.

Everything seems to work fine. LoRaWAN connection, BLE and WiFi counting. Downlinks. All works.

However, the GPS continues to report only 0,0 - whatever I do. Have been trying to use internal GPS antenna and external antenna. It just looks like the GPS is not properly read and/or activated. I get no errors at all. Can not see any debugging information to give me an idea where to look at what is going wrong. 

Does someone have any idea what is missing here?

cyberman54, 2021-01-07:
What message do you get on serial console while device is starting up:
`Quectel L76 GPS chip found` or `Quectel L76 GPS chip not found on i2c bus, bus error <X>`

And if not found, what is printed as "X"?

tjvandam, 2021-01-07:
This is the serial output I get:

`> [I][gpsread.cpp:56] gps_init(): Quectel L76 GPS chip found`

If have tried to put the device outside with clear sky view for 30 minutes. It continues to report 0,0 on port 4 when I send a port 2 downlink with 0x84.

Any other ideas for debugging?

cyberman54, 2021-01-07:
Do you see any errors or reboots in the serial log?

tjvandam, 2021-01-07:
No errors or reboots. All looks good.

cyberman54, 2021-01-07:
I checked it. Found a bug, that prevents the gps task starting if a i2c connected gps is used. Pytrack is one of the rare boards which is using i2c gps chip. I [commited a fix](https://github.com/cyberman54/ESP32-Paxcounter/pull/713/commits/dee4d825a06f71a9566ee1d95b625e8ecde5d15a). Please recompile current master software and try again.

cyberman54, 2021-01-08:
Meanwhile i could test the bugfix it with a LoPy (old one, a non 4) on a (old v1.1) Pytrack board. It works:

```
{
  "altitude": 37,
  "hdop": 2.18,
  "latitude": 52.xxxxx,
  "longitude": 13.xxxxx,
  "sats": 5
}
```

Note: it took ~15 minutes until the Quectel L76 gps chip got a fix after coldstart in an urban area. While it has no fix, no gps data is transmitted by paxcounter software.

tjvandam, 2021-01-08:
Works on my side also now. Thanks for the quick fix!

cyberman54, 2021-01-08:
@tjvandam Can you confirm that it works on a Pytrack 2.0 board?

closed 2021-01-08 by cyberman54

tjvandam, 2021-01-08:
Yes I can confirm that.