cyberman54/ESP32-Paxcounter #714: PAX counter drops to 0 after a short while

HouzuoGuo, 2021-01-08:
Good day!

According to my observations made on this TTGO T-Beam v0.7, the PAX counter (WiFi + BLE combined) always drops to 0 after the microcontroller has been running for couple of minutes (in some cases hours).

[image]

Points of interest:

- Shortly before 21:23, I manually reset the microcontroller, hence the reading dropped to 0 very briefly.
- At around 21:40 (~15 minutes of uptime), the microcontroller's PAX counter reading dropped to 0, and the reading of 0 lasted for couple of hours until next morning.
- Microcontroller did not appear to experience other faults, as evident from the its continuous transmission of GPS data over LoRaWAN, as illustrated on the widgets.
- At around 07:00, the PAX counter came back to life due to triggering of brownout detector.

Here are the latest log entries, earlier entries were lost from vscode console:

<details><summary>Logs</summary>
<pre>

06:32:37.445 > [D][lorawan.cpp:399] myEventCallback(): TXCOMPLETE    10
06:32:37.848 > [I][lorawan.cpp:214] lora_send(): 4 byte(s) sent to LORA
06:32:43.714 > [D][cyclic.cpp:30] doHousekeeping(): Heap: Free:110756, Min:102292, Size:236388, Alloc:90656, StackHWM:2312
06:32:43.715 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
06:32:43.715 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 2
06:32:43.715 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
06:32:43.715 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2284 bytes left | Taskstate = 1
06:32:43.715 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1480 bytes left | Taskstate = 2
06:32:43.729 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2304 bytes left | Taskstate = 1
06:32:43.729 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
06:33:13.715 > [D][cyclic.cpp:30] doHousekeeping(): Heap: Free:110756, Min:102292, Size:236388, Alloc:90656, StackHWM:2312
06:33:13.716 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
06:33:13.716 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 2
06:33:13.716 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
06:33:13.716 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2284 bytes left | Taskstate = 1
06:33:13.716 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1480 bytes left | Taskstate = 2
06:33:13.764 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2304 bytes left | Taskstate = 1
06:33:13.764 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
06:33:13.764 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 1
06:33:13.764 > [I][senddata.cpp:117] sendData(): Counter cleared
06:33:13.764 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 4
06:33:13.764 > [W][lorawan.cpp:320] lora_enqueuedata(): LORA sendqueue is full
06:33:15.961 > [D][lorawan.cpp:399] myEventCallback(): TXSTART       10
06:33:17.279 > [D][lorawan.cpp:399] myEventCallback(): RXSTART       10
06:33:18.280 > [D][lorawan.cpp:399] myEventCallback(): RXSTART       10
06:33:18.522 > [D][lorawan.cpp:399] myEventCallback(): TXCOMPLETE    10
06:33:18.531 > [I][lorawan.cpp:214] lora_send(): 4 byte(s) sent to LORA
06:33:43.715 > [D][cyclic.cpp:30] doHousekeeping(): Heap: Free:110756, Min:102292, Size:236388, Alloc:90656, StackHWM:2312
06:33:43.716 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
06:33:43.716 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 2
06:33:43.716 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
06:33:43.716 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2284 bytes left | Taskstate = 2
06:33:43.716 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1480 bytes left | Taskstate = 2
06:33:43.743 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2304 bytes left | Taskstate = 1
06:33:43.743 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
06:33:48.926 > [D][lorawan.cpp:399] myEventCallback(): TXSTART        9
06:33:50.247 > [D][lorawan.cpp:399] myEventCallback(): RXSTART        9
06:33:51.247 > [D][lorawan.cpp:399] myEventCallback(): RXSTART        9
06:33:51.489 > [D][lorawan.cpp:399] myEventCallback(): TXCOMPLETE     9
06:33:51.615 > [I][lorawan.cpp:214] lora_send(): 13 byte(s) sent to LORA
06:34:13.716 > [D][cyclic.cpp:30] doHousekeeping(): Heap: Free:110756, Min:102292, Size:236388, Alloc:90656, StackHWM:2312
06:34:13.717 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
06:34:13.717 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 2
06:34:13.717 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
06:34:13.717 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2284 bytes left | Taskstate = 2
06:34:13.717 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1480 bytes left | Taskstate = 2
06:34:13.750 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2304 bytes left | Taskstate = 1
06:34:13.750 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
06:34:13.750 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 1
06:34:13.750 > [I][senddata.cpp:117] sendData(): Counter cleared
06:34:13.750 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 4
06:34:21.893 > [D][lorawan.cpp:399] myEventCallback(): TXSTART       10
[... 936 lines cut ...]
07:06:23.617 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
07:06:23.617 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 1
07:06:23.617 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
07:06:23.617 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2392 bytes left | Taskstate = 2
07:06:23.617 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1492 bytes left | Taskstate = 2
07:06:23.647 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2316 bytes left | Taskstate = 1
07:06:23.647 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
07:06:23.647 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 1
07:06:23.647 > [I][senddata.cpp:117] sendData(): Counter cleared
07:06:23.647 > [D][senddata.cpp:13] SendPayload(): sending Payload for Port 4
07:06:24.857 > [D][lorawan.cpp:399] myEventCallback(): TXSTART        2
07:06:26.259 > [D][lorawan.cpp:399] myEventCallback(): RXSTART        2
07:06:27.258 > [D][lorawan.cpp:399] myEventCallback(): RXSTART        2
07:06:27.501 > [D][lorawan.cpp:399] myEventCallback(): TXCOMPLETE     2
07:06:27.670 > [I][lorawan.cpp:214] lora_send(): 4 byte(s) sent to LORA
07:06:53.618 > [D][cyclic.cpp:30] doHousekeeping(): Heap: Free:110756, Min:102324, Size:236392, Alloc:90656, StackHWM:2312
07:06:53.618 > [D][cyclic.cpp:33] doHousekeeping(): IRQhandler 2312 bytes left | Taskstate = 0
07:06:53.618 > [D][cyclic.cpp:36] doHousekeeping(): MACprocessor 2392 bytes left | Taskstate = 2
07:06:53.618 > [D][cyclic.cpp:38] doHousekeeping(): Rcommand interpreter 2632 bytes left | Taskstate = 2
07:06:53.618 > [D][cyclic.cpp:41] doHousekeeping(): LMiCtask 2392 bytes left | Taskstate = 2
07:06:53.618 > [D][cyclic.cpp:44] doHousekeeping(): Lorasendtask 1492 bytes left | Taskstate = 2
07:06:53.632 > [D][cyclic.cpp:48] doHousekeeping(): Gpsloop 2316 bytes left | Taskstate = 1
07:06:53.632 > [D][cyclic.cpp:67] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
</pre>
</details>

HouzuoGuo, 2021-01-08:
As a temporary measure, i'm going to reset the microcontroller if there are too many consecutive readings of 0 on the pax counter:

https://github.com/HouzuoGuo/ESP32-Paxcounter/commit/c18ff869f1f5d10d0c8ed2cd2c65fd78b130c59d

See how that goes..

HouzuoGuo, 2021-01-08:
The readings on OLED further confirm the observations made from the dashboard, both WiFi and BLE counters read 0, while the ID of wifi channel being scanned keeps on rolling - it just happens to read 0 at the moment the picture was taken.

[image]

cyberman54, 2021-01-08:
Sorry that you encounter problems running this software.

Zero values could be a result of unsolicited board resets. 
To ensure that there a no resets, please check uptime value from your board (rcommand 0x81) after you encountered zero values.

cyberman54, 2021-01-08:
Your log shows:

- the board shows brownout resets => Check your power supply. It may be to weak in peak power.
- the LORA sendqueue is full => Check if your board joins the LORA network. If so, reduce the amoung of data to be sent (e.g. switch off GPS messages, or increase send cycle.)

HouzuoGuo, 2021-01-08:
Thank you, i’ll dig deeper and look for the cause of a full lora send queue. According to TTN console, the transmitter indeed was sending consecutive readings of 0 PAX.

In the meanwhile, here’s the dashboard:
https://admin.tago.io/public/dashboard/5ff7361709890a0027afb78f/b97d8a56-926e-4c8b-b0cc-46d8633eced3

My temporary, auto-reset patch is running on the transmitter right now, i want to see how often does the auto-reset have to happen.

cyberman54, 2021-01-08:
Check uptime value of the board, to ensure it has not unsolicited resets.
Check frame counter in TTN console, to ensure there are no rejoins.

I never saw this issue, and no other user reported this yet.

So, if this is a bug in the code, i will be very interested to find out the root cause.

cyberman54, 2021-01-08:
@HouzuoGuo Can you, please, post you paxcounter.conf here, thanks.

cyberman54, 2021-01-08:
and one more hint: in vscode, write log file with pio device monitor -f log2file

cyberman54, 2021-01-08:
and another hint: if you're using TTN as LORAWAN network, you may activate data storage integration in your TTN application. Then retrieve the raw payload from the database to check.

HouzuoGuo, 2021-01-10:
Thanks for the hints!

Here's the branch with my configuration files, couple of cosmetic changes made for the OLED display, and couple more tweaks including the auto-reset patch:

https://github.com/HouzuoGuo/ESP32-Paxcounter/commits/master

Here is the paxcounter.conf:

https://github.com/HouzuoGuo/ESP32-Paxcounter/blob/master/src/paxcounter.conf

(BTW, the ABP keys in https://github.com/HouzuoGuo/ESP32-Paxcounter/blob/master/src/loraconf.h are no longer current)

According to the data of the past two days, as plotted on the dashboard, the auto-reset patch seems to have kicked in 15 times, the interval in between is rather random:

[image]

Next, I'll collect the log output in a text file and see if there are more clues.

HouzuoGuo, 2021-01-10:
On a related note, do you like this new GPS info display, with more details such as UTC time and altitude? If so, I'll make a PR for it:

[image]

cyberman54, 2021-01-10:
Thanks for your improvements here. Yes, your are welcome to redesign screen layout. If you make changes to the layout, keep in mind that same layout must work on TFT displays (e.g. TTGO T-DISPLAY or M5 stack), too.

But first we should keep focus on this weird issue with zeroes in the counter. I have no clue what could cause this, and i can not reproduce this issue on any of my boards, including a T-Beam v0.7 which runs for days now.

Thus, the issue may depend on local context, e.g. power supply / timezone / Lorawan network or parameters / board hardware ...

HouzuoGuo, 2021-01-11:
I suspected that the power supply from my laptop's USB port may have been unstable, after digging around in the source code, I managed to:

- Prevent the LoRa outgoing packet queue from filling up by increasing the interval between sending payloads: https://github.com/HouzuoGuo/ESP32-Paxcounter/commit/f6ab806266462b04b71f90b5a16ba798dc60788a
- Send internal battery voltage reading to TTN: https://github.com/HouzuoGuo/ESP32-Paxcounter/commit/005fc2c3ce56d35408f97ab01a35689c44649572

In addition to couple of cosmetic changes, and automatically resetting the microcontroller if there have been consecutive readings of PAX number 0 for 10 minutes: https://github.com/HouzuoGuo/ESP32-Paxcounter/commit/c09a52cef8d72f64b3e40ae2560ec56e5290c794

My T-Beam is running the code from that forked master branch, the LoRa outgoing packet queue is no longer filling up, the number of queues messages often reads 0 and it is always below 4. Although according to both OLED display and TTN application data console, the PAX counter reading still drops to 0 (and stays at 0) quite often:

[image]

Next round, I'll leave the board connected to laptop and collect the serial output into a file.

cyberman54, 2021-01-11:
Do you have a second board to test? Maybe it's hardware problem.

cyberman54, 2021-01-11:
To check power supply, please dismount 18650 battery from T-Beam while testing / logging, to reduce USB current.

HouzuoGuo, 2021-01-11:
That's a good point, perhaps the board itself is faulty.
I don't have another T-Beam v0.7, however there are two v1.1 boards coming in couple of days via mail.
Let me open a new issue when this problem happens on another board.

closed 2021-01-11 by HouzuoGuo

cyberman54, 2021-01-25:
As is see on your dashboard, you got it up and running now without zero drops? What did you change?

HouzuoGuo, 2021-01-25:
@cyberman54 hehe - you were totally right, the ttgo t-beam v0.7 board must have been faulty.

I put it away in favour of a newly arrived t-beam v1.1, along with some bme280 sensor boards. This new t-beam works far better and pax counter has been working fine.