cyberman54/ESP32-Paxcounter #724: Problem with wifi TCP/IP MQTT

HelloWar75, 2021-01-18:
I have problem with wifi tcp/ip for mqtt.

My board is heltec lora32 v2.

`
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
04:39:21.349 > ET),boot:0x1[I][reset.cpp:81] do_after_reset(): Starting Software v2.2.0, runmode powercycle
04:39:21.349 > [I][main.cpp:159] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 8MB external Flash
04:39:21.349 > [I][main.cpp:161] setup(): Internal Total heap 267080, internal Free Heap 240680
04:39:21.349 > [I][main.cpp:167] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2.3-14-gd3e562907
04:39:21.349 > [I][main.cpp:169] setup(): Flash Size 8388608, Flash Speed 40000000
04:39:21.349 > [I][main.cpp:171] setup(): Wifi/BT software coexist version 1.1.7
04:39:21.349 > [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(4)=0, scl(15)=0
04:39:21.349 > [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
04:39:21.349 > [I][configmanager.cpp:98] loadConfig(): Loading device configuration from NVRAM...
04:39:21.349 > [I][configmanager.cpp:123] loadConfig(): Runtime configuration v2.2.0 loaded
04:39:21.349 > [I][i2c.cpp:44] i2c_scan(): Starting I2C bus scan...
04:39:21.349 > [I][i2c.cpp:78] i2c_scan(): 0 I2C device(s) found
04:39:21.525 > [I][main.cpp:259] setup(): Starting LED Controller...
04:39:21.580 > [I][main.cpp:294] setup(): Starting MAC processor...
04:39:21.580 > [I][macsniff.cpp:54] macQueueInit(): MAC processing queue created, size 600 Bytes
04:39:21.580 > [I][main.cpp:298] setup(): Starting rcommand interpreter...
04:39:21.580 > [I][rcommand.cpp:493] rcmd_init(): Rcommand send queue created, size 55 Bytes
04:39:22.546 > E (3459) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000
04:39:23.546 > E (4460) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0x0000
04:39:23.557 > E (4461) emac: Initialise PHY device Timeout
04:39:23.568 > [E][ETH.cpp:101] begin(): esp_eth_enable error: -1
04:39:23.568 > [I][mqttclient.cpp:36] mqtt_init(): MQTT send queue created, size 510 Bytes
04:39:23.572 > [I][mqttclient.cpp:38] mqtt_init(): Starting MQTTloop...
04:39:23.577 > I (4481) wifi: wifi driver task: 3ffbbe20, prio:23, stack:3584, core=0
04:39:23.583 > I (8994) wifi: wifi firmware version: 7997e4b
04:39:23.587 > I (8996) wifi: config NVS flash: disabled
04:39:23.595 > I (9000) wifi: config nano formating: disabled
04:39:23.595 > I (9004) wifi: Init dynamic tx buffer num: 32
04:39:23.599 > I (9008) wifi: Init data frame dynamic rx buffer num: 32
04:39:23.604 > I (9013) wifi: Init management frame dynamic rx buffer num: 32
04:39:23.611 > I (9019) wifi: Init management short buffer num: 32
04:39:23.614 > I (9024) wifi: Init static rx buffer size: 1600
04:39:23.619 > I (9028) wifi: Init static rx buffer num: 16
04:39:23.627 > I (9032) wifi: Init dynamic rx buffer num: 32
04:39:23.627 > I (9036) wifi: set country: cc=BR␀ schan=1 nchan=13 policy=1
04:39:23.632 >
04:39:23.632 > W (9041) wifi: ap start fail
04:39:23.635 >
04:39:23.635 > I (9044) wifi: Set ps type: 0
04:39:23.643 >
04:39:23.643 > [I][main.cpp:426] setup(): Starting Wifi...
04:39:23.689 > I (9109) wifi: mode : null
04:39:23.692 > I (9110) wifi: ic_enable_sniffer
04:39:23.695 > [I][main.cpp:441] setup(): Starting Interrupt Handler...
04:39:23.700 > [I][main.cpp:469] setup(): Starting Timers...
04:39:23.704 > [I][main.cpp:527] setup(): Features: VEXT OLED LED OTA MQTT FILTER PLAIN WIFI BTN_PD
04:39:43.566 > [I][mqttclient.cpp:54] mqtt_connect(): MQTT name is paxcounter_ce154e9a
04:39:43.572 > [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for public.cloud.shiftr.io
04:39:43.579 > [I][mqttclient.cpp:61] mqtt_connect(): Could not resolve public.cloud.shiftr.io
04:39:53.728 > [I][senddata.cpp:117] sendData(): Counter cleared
04:40:03.576 > [I][mqttclient.cpp:54] mqtt_connect(): MQTT name is paxcounter_ce154e9a
04:40:03.582 > [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for public.cloud.shiftr.io
04:40:03.589 > [I][mqttclient.cpp:61] mqtt_connect(): Could not resolve public.cloud.shiftr.io
04:40:23.585 > [I][mqttclient.cpp:54] mqtt_connect(): MQTT name is paxcounter_ce154e9a
04:40:23.592 > [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for public.cloud.shiftr.io
04:40:23.599 > [I][mqttclient.cpp:61] mqtt_connect(): Could not resolve public.cloud.shiftr.io
04:40:23.729 > [I][senddata.cpp:117] sendData(): Counter cleared
04:40:43.596 > [I][mqttclient.cpp:54] mqtt_connect(): MQTT name is paxcounter_ce154e9a
04:40:43.602 > [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for public.cloud.shiftr.io
04:40:43.609 > [I][mqttclient.cpp:61] mqtt_connect(): Could not resolve public.cloud.shiftr.io
04:40:53.730 > [I][senddata.cpp:117] sendData(): Counter cleared
04:41:03.604 > [I][mqttclient.cpp:54] mqtt_connect(): MQTT name is paxcounter_ce154e9a
04:41:03.611 > [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for public.cloud.shiftr.io
[... 389 lines cut ...]
board = esp32dev
board_build.partitions = min_spiffs.csv
upload_speed = ${common.upload_speed}
platform = ${common.platform_espressif32}
lib_deps = ${common.lib_deps_all}
build_flags = ${common.build_flags_all}
upload_protocol = ${common.upload_protocol}
extra_scripts = ${common.extra_scripts}
monitor_speed = ${common.monitor_speed}
monitor_filters = 
	time
	esp32_exception_decoder
	default

[env:ota]
upload_protocol = custom

[env:usb]
upload_protocol = esptool

[env:dev]
upload_protocol = esptool
build_type = debug
platform = https://github.com/platformio/platform-espressif32.git#develop
`

cyberman54, 2021-01-18:
MQTT over Wifi is not (yet) supported, you need a PHY with Ethernet.
So you can't use this option with a Heltec V2 LORA board.
Use the MQTT connector of your LORAWAN network server instead.

closed 2021-01-18 by cyberman54

dsandels, 2023-05-12:
Hi,

I have a ESP32-Paxcounter LoRa32 Bluetooth WiFi Module 915MHZ module. How do I find the APP EUI and the APP Key for operation on the Helium network ?

Thanks !

cyberman54, 2023-05-12:
@dsandels Please open new thread in github tab "discussion" and post your question there.

dsandels, 2023-05-14:
Thanks for the advice !Unfortunately, I am totally new to GitHub and cannot find the “discussion tab”. Where do I go from here ?Thanks in advance !Sent from my iPhoneDavid Sandel iNeighborhoods.usSt. Louis, MO.Austin, TX.EinsteinDesigns.center314-435-3658