cyberman54/ESP32-Paxcounter #736: The Things Stack V3 payload decoder integration

Drasek, 2021-02-02:
Hello Cyberman54,

the payload decoder in the things stack v3 works a bit different than the one in v2, so i do provide both decoders (packed and plain) further down. I made them V3 compatible and did harmonize some of the valuenames (battery -> voltage, temp -> cputemp) between both versions. I did include the converting faunctions into the decoder as well, because there is only one function in the backend in v3. I named them "packed_decodeUplink.js" and "plain_decodeUplink.js" after the new function, maybe you could commit them to this repo:

Drasek, 2021-02-02:
`// Decoder for device payload encoder "PACKED" - "packed_decodeUplink.js"
// copy&paste to TTN Console V3 -> Applications -> Payload formatters -> Uplink -> Javascript
// modified for The Things Stack V3 by Caspar Armster, dasdigidings e.V.

function decodeUplink(input) {
    var data = {};

    if (input.fPort === 1) {
        // only wifi counter data, no gps
        if (input.bytes.length === 2) {
            data = decode(input.bytes, [uint16], ['wifi']);
        }
        // wifi + ble counter data, no gps
        if (input.bytes.length === 4) {
            data = decode(input.bytes, [uint16, uint16], ['wifi', 'ble']);
        }
        // combined wifi + ble + SDS011
        if (input.bytes.length === 8) {
            data = decode(input.bytes, [uint16, uint16, uint16, uint16], ['wifi', 'ble', 'PM10', 'PM25']);
        }
        // combined wifi counter and gps data, used by https://opensensemap.org
        if (input.bytes.length === 10) {
            data = decode(input.bytes, [latLng, latLng, uint16], ['latitude', 'longitude', 'wifi']);
        }
        // combined wifi + ble counter and gps data, used by https://opensensemap.org
        if (input.bytes.length === 12) {
            data = decode(input.bytes, [latLng, latLng, uint16, uint16], ['latitude', 'longitude', 'wifi', 'ble']);
        }
        // combined wifi counter and gps data
        if (input.bytes.length === 15) {
            data = decode(input.bytes, [uint16, latLng, latLng, uint8, hdop, altitude], ['wifi', 'latitude', 'longitude', 'sats', 'hdop', 'altitude']);
        }
        // combined wifi + ble counter and gps data
        if (input.bytes.length === 17) {
            data = decode(input.bytes, [uint16, uint16, latLng, latLng, uint8, hdop, altitude], ['wifi', 'ble', 'latitude', 'longitude', 'sats', 'hdop', 'altitude']);
        }
        
        data.pax = 0;
        if ('wifi' in data) {
            data.pax += data.wifi;
        }
        if ('ble' in data) {
            data.pax += data.ble;
        } 
    }

    if (input.fPort === 2) {
        // device status data
        if (input.bytes.length === 17) {
            data = decode(input.bytes, [uint16, uptime, uint8, uint32, uint8, uint8], ['voltage', 'uptime', 'cputemp', 'memory', 'reset0', 'reset1']);
        }
    }

    if (input.fPort === 3) {
        // device config data      
        data = decode(input.bytes, [uint8, uint8, int16, uint8, uint8, uint8, uint8, bitmap1, bitmap2, version], ['loradr', 'txpower', 'rssilimit', 'sendcycle', 'wifichancycle', 'blescantime', 'rgblum', 'flags', 'payloadmask', 'version']);
    }

    if (input.fPort === 4) {
        // gps data      
        if (input.bytes.length === 8) {
            data = decode(input.bytes, [latLng, latLng], ['latitude', 'longitude']);
        } else {
            data = decode(input.bytes, [latLng, latLng, uint8, hdop, altitude], ['latitude', 'longitude', 'sats', 'hdop', 'altitude']);
        }
    }

    if (input.fPort === 5) {
        // button pressed      
        data = decode(input.bytes, [uint8], ['button']);
[... 233 lines cut ...]
            return prev;
        }, {});
};

if (typeof module === 'object' && typeof module.exports !== 'undefined') {
    module.exports = {
        uint8: uint8,
        uint16: uint16,
        uint32: uint32,
        int8: int8,
        int16: int16,
        int32: int32,
        uptime: uptime,
        float: float,
        ufloat: ufloat,
        pressure: pressure,
        latLng: latLng,
        hdop: hdop,
        altitude: altitude,
        bitmap1: bitmap1,
        bitmap2: bitmap2,
        version: version,
        decode: decode
    };
}`

hekopath, 2021-02-02:
Thank you for writing the new decoders.
Is there a reference for the mentioned hamonisation of valuenames?
In most cases the payload temperature will contain ambient and not the cpu temperature. This can be misleading

Drasek, 2021-02-02:
Its just 2 values, one is "battery" (used in the plain decoder) which is now "voltage" (used in the packed decoder and now in both) and if you have a look in the if statement for port === 2 (plain decoder) there is the value named "temp" while it is the "cputemp" (packed decoder). It gets transmitted together with "uptime", "memory" - so we talk about the cpu or board temperature in this case and not the ambient temperature.

cyberman54, 2021-02-02:
Thanks for your contribution! Unfortunately i am not yet into v3, need to find some time to dive in. Do your decoders work for v2 and v3, or is this a v3 only topic?

If they run with v2, shall we do a breaking change? Or offer them as additional option?

hekopath, 2021-02-02:
> Its just 2 values, one is "battery" (used in the plain decoder) which is now "voltage" (used in the packed decoder and now in both) and if you have a look in the if statement for port === 2 (plain [...]

Thank you for clarification.
I did not realize that there is a difference between the two versions.
When I first read it, I related the "harmonization" to the v3 stack.

Drasek, 2021-02-02:
The decoder only work for V3 (thats why i changed the name), i would add them into "src/TTN". In the first few lines inside the decoder, in the comment, there is a description where to put them on the v3 console (website). I can share a screenshot...

[image]

The new decoding function without content:
`function decodeUplink(input) {
  return {
    data: {
      bytes: input.bytes
    },
    warnings: [],
    errors: []
  };
}`

Now you get one object with port and bytes instead of two variables (v2) and as return value you ca give warnings and errors as well. The functions i provide follow this new code and while i was at it is i ereased the differences between the plain and packed output. Ohh and i included one more decoding option for people who use the sds011 particle matter sensor with the paxcounter. the decoding from payloads containing this information was missing.

hekopath, 2021-02-02:
If TTN stay at their announcment:
"Warning The Things Network is shutting down v2 clusters later this year. Start moving your applications and devices to a v3 cluster!"
The breaking change will be forced by TTN.
As far as i can see there are several services/integrations (TTN Mapper for Example) which does not work with v3 at the moment.
Probably a new branch is an option. Espacially if there are more changes for the v3 specific documentation.

cyberman54, 2021-02-02:
Added to new folder /src/TTNv3.
Thanks for the contributing!

closed 2021-02-02 by cyberman54