cyberman54/ESP32-Paxcounter #740: Strange deep-sleep behavior

StefanStieger, 2021-02-10:
I am thrilled with paxcounter - cool stuff. I have already tried out a lot with it and also tested it with several different boards. Now I want to use the paxcounter for long-term counting, i.e., paxcounter should count for half a minute and then go to deep-sleep for - let's say - an hour. 

This is my configuration:
TTGO T2 v1.6 (no lora, but SD card)
I use the lolin32lite.h file as suggested in another thread because board has no LoRa. Because the display is SSD1331, I disabled it - no problem so far. 
Count is cumulative and will be stored on SD card and written to RTC memory before going to deep sleep.
This all works fine, BUT only when the device is connected to USB. As soon as I unplug it, and the device runs only on battery, the board makes one cycle, goes to deep-sleep and does not wake up again. 
And now the very strange thing. When I set the SLEEPCYCLE to 30 seconds (value = 15) then it works, also in battery mode but not with higher values i.e., seconds for SLEEPCYCLE.

First idea was that maybe my battery (LiPo 3.7 V 800 mAh) has some shut-off procedure because of too little power used by the board but changing to a 18650 cell shows the same behavior.

Any ideas? 

Best,
Stefan

cyberman54, 2021-02-11:
please try disable SD card write by disabling the SD compile option, and compare behaviour, then post result here. (SD write was not yet fully tested in conjunction with deep sleep)

StefanStieger, 2021-02-11:
Ok, tested. Unfortunately same behavior. When on usb, everything works fine. When on battery, one cycle (meanwhile I know the exact green led flashlight order) and then nothing.

My first guess was it probably has something to do with the battery, because when connected to usb everything works (even writing the data to the SD-card) and the serial monitor does not show any anomalies (even when using debug_level = 5). But when on battery it seems only to work for a minute or so because with the SLEEPCYCLE = 15 (=30 seconds) it works even on battery, but when this value is larger, the problem occurs.

Thankful for any hints.

Best,
Stefan

cyberman54, 2021-02-11:
So, the issue is, board does not wake up after deep sleep in battery mode?
Is it [this](https://github.com/LilyGO/TTGO-T2-SSD1331-SD) type of board, i.e. does the schematic in that repo apply?

StefanStieger, 2021-02-11:
Yes, exactly. T2 V1.6.

cyberman54, 2021-02-11:
Board description says the device shall have a button on IO36.
Please try to configure this button in hal file by setting `HAS_BUTTON (36)`, then try if manual waking up the device from deep sleep by pressing the button works, while device runs on battery.

StefanStieger, 2021-02-11:
Excellent idea. Tested and indeed wakes up the board from deep-sleep when pressed but only when on USB. During battery operation, the button does not wake up the board, no led flashes - just dead. This implies that the strange behavior might has nothing to do with deep-sleep but something else.

Meanwhile, I have it also tried a TTGO T5 e-ink (v2.3)
[https://github.com/Xinyuan-LilyGO/T5-Ink-Screen-Series](https://github.com/Xinyuan-LilyGO/T5-Ink-Screen-Series)
Again, deactivated display and adjusted PINs for SD-card. Everything works fine - and now the good news - also for longer sleep-times at least for 60 seconds. Currently testing with much longer sleep-times. 
This means that the strange issue might have something todo with the T2 board itself.

My problem is, that power consumption is very important for me. T5 takes 120mA and T2 only 80mA so I would prefer T2. Or can you recommend any other board with SD-card which can be used as a paxcounter but has very low power consumption during scanning?

Best,
Stefan

cyberman54, 2021-02-11:
Power consumption by ESP32 while scanning is far higher than consumption by peripherals of the board, around 0,8 watt. So it's not board hardware specific. You can reduce the ESP32 power consumption while scanning by shortening scan times only. Either by using a (deep) sleep cycle, or - for BLE scanning - by adjusting BLE scan windows (see paxcounter.conf).

Board hardware becomes relevant when it comes to power consumption in deep sleep mode. Best solution is to select boards with a PMU chip, which allows software-driven cutoff of power hungry board peripherals, like USB-UART converter, display, gps, etc. Not too bad is the TTGO T-Beam v1.1, it has AXP192 PMU chip, supported by paxcounter.

Ask google for ESP32 ultra low power, e.g. look here
https://community.hiveeyes.org/t/low-power-esp32-hardware-and-software/538/8

cyberman54, 2021-02-11:
Did you try a Olimex board?

https://www.olimex.com/Products/IoT/ESP32/

StefanStieger, 2021-02-11:
Thanks for the hints. I have a TTGO T-Beam v1.1 laying around - will try that one although rather bulky.  Also thanks for the Olimex board - wasn't aware of those ones. Will have a closer look. My main reason for TTGO T2 was that it is very small and handy. I want to build a mobile paxcounter for a scientific study. Therefore it should be as small as possible so that participants can keep it in a pocket or bag easily. Nevertheless, thanks for your support. Very much appreciated.

cyberman54, 2021-02-11:
What about this one?
https://zerforschung.org/p/cov32-artikel/messgeraet.jpg

closed 2021-02-11 by cyberman54

StefanStieger, 2021-02-11:
Looks interesting but what is it? Counter?

cyberman54, 2021-02-11:
Is a very similary device. Same approach, but other use case. You can ask them at zerforschung.org which board they used (probably purchaed on Aliexepress) and what deep sleep consumption they achieved.