cyberman54/ESP32-Paxcounter #741: Review LoRa time sync settings

mloeschner, 2021-02-14:
It seems to me that time snyc via LoRa is always active, since in https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/lmic_config.h#L32 `LMIC_ENABLE_DeviceTimeReq` is set to `1`.

The code to evaluate the actual setting is in https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/lorawan.cpp#L16-L20.

I removed the definition in `lmic_config.h` but I'm not sure if I had overseen anything.

cyberman54, 2021-02-14:
`LMIC_ENABLE_DeviceTimeReq` in _lmic_config.h_ controls if the LMIC stack will process `DeviceTimeRequests` on MAC layer, or not. It's a compiler option to reduce code size, if time sync is not needed by the node.

You can keep this compiler option enabled, even if your LORAWAN network server does not support DeviceTimeRequests, like TTN v2, if you don't care about code size. Since ESP32 provides huge flash ROM size, paxcounter enables the compiler option.

User control, if network time is actually used or not, is given in _paxcounter.conf_ by `#define TIME_SYNC_LORAWAN`.
Set this to 0, if you don't need network time, or if your LORAWAN network server does not provide network time.

So, i don't think this is a bug, or could be an issue causing any problems.

mloeschner, 2021-02-15:
Thanks for the clarification.

closed 2021-02-15 by mloeschner