cyberman54/ESP32-Paxcounter #749: Wifi scan cycle too fast

mikebreeuwer, 2021-02-25:
Somehow my board scans all wifi channels in under a second, and finds only 3 to 5 items (running airodump on my laptop finds over 20 randomized mac addresses on average).
The strange thing is that changing WIFI_CHANNEL_SWITCH_INTERVAL (from 50 to 100 or even 200) in paxcounter.conf does not show any slow down at all. Terminal output shows no errors.
I compiled the latest zip from github and I'm running on a 'TOOGOO' board (https://www.amazon.de/gp/product/B07NKY9PNW) and even though it has TTGO printed on the back it's not a TTGO Lora32 v1 but it seems to be a Heltec v1. The LED's (it has at least a blue and a red one) don't light up when running.
Since I have not configured TTN yet I've compiled without Lora active. Compiling without BLE does not change the fast scanning bahaviour.
Compiling itself is without errors, only this warning:
BLERemoteService.cpp:246:89: warning: parameter 'pCharacteristicMap' set but not used [-Wunused-but-set-parameter]

cyberman54, 2021-02-25:
Thank you for reporting this behaviour, and sorry that you had trouble running this software.

You indeed spotted a _major_ bug, making wifi channel rotation 10 times too fast!

So you probably could not recognize the impact after you modified the `WIFI_CHANNEL_SWITCH_INTERVAL` setting.
The bug came in on 10th Dec 2020 with this [commit](https://github.com/cyberman54/ESP32-Paxcounter/commit/28d30493fde556cee758891049d79ea313a35434#diff-2db3344377dee921191944995ae67d7c14ca62cedad71d8c5c57e4ad04d65dd1).
I fixed it in master.

/cc @spmrider @FlorianLudwig

closed 2021-02-25 by cyberman54

embeeweezer, 2021-02-26:
Danke sehr & keep up the good work!

*Studio Mike Breeuwer* | Amsterdam
+31 6 8155 8787