cyberman54/ESP32-Paxcounter #752: Confusing definitions of `BUTTON_PULLUP` and `BUTTON_PULLDOWN`

lenaschimmel, 2021-03-05:
The way the button hardware is defined seems ambiguous to me. In [`main.c`](https://github.com/cyberman54/ESP32-Paxcounter/blob/011c5390356e0f082bca870c600dc5388885f475/src/main.cpp#L494) there is a check for `BUTTON_PULLUP`, while [`button.c`](https://github.com/cyberman54/ESP32-Paxcounter/blob/011c5390356e0f082bca870c600dc5388885f475/src/button.cpp#L14) checks for `BUTTON_PULLDOWN`:

So it seems to me that one should always define `BUTTON_PULLUP` or `BUTTON_PULLDOWN`. When defining both, or defining none, the output on startup and the actual behavior do not match, but there's no error or warning. Five of the `hal/*.h`headers define `BUTTON_PULLUP`, none of the files define `BUTTON_PULLDOWN`.

I noticed that the button functionality uses the Arduino lib [SimpleButton](https://github.com/spacehuhn/SimpleButton#push-button-setup) by @spacehuhn, which has:

 *`Button`, [which needs an external pull up or pull down resistor](https://www.arduino.cc/reference/de/language/functions/digital-io/pinmode/)
 *`ButtonPullup`, [which enables the integrated pull up resistor](https://github.com/spacehuhn/SimpleButton#push-button-setup)

So it would be consistent to use only the definition `BUTTON_PULLUP` and get rid of `BUTTON_PULLDOWN`.

_(I can also make a PR for this, but since I'm completely new to Arduino and ESP32 development, first I'd like to hear if I'm on the right track.)_

cyberman54, 2021-03-05:
Yes, that's right, we should get rid of `BUTTON_PULLDOWN` here.
You're welcome to create a PR for this.

Generally, the SimpleButton Arduino lib should be changed by a more popular, better maintained lib.
Any suggestions would be welcome.

cyberman54, 2022-08-04:
fixed by https://github.com/cyberman54/ESP32-Paxcounter/pull/884

closed 2022-08-04 by cyberman54