cyberman54/ESP32-Paxcounter #766: Need support with downlink Remote Control/Battery

SionHuws99, 2021-03-23:
Hi,

I am brand new to both TTN and was hoping to get some help on the following.

I have uploaded the paxcounter source code onto a TTGO T3_V1.6 board and connected to my TTN account, it works well so far although when connecting the board to 4 AA after a few hours the screen turns black only leaving a red light. I was guessing this was due to it using up all of the batteries.

If this is the case I was hoping I could turn the display off on the sensor and set it to sleep in 10 minute intervals where it would count, send the data and then sleep again for 10 minutes.

I noticed on the documentation there is instructions for remote control using the TTN downlink, but I am un-sure how to enter the correct numbers to trigger the display or set the sleep cycle.

I am struggling to resolve this and couldn't find anything online so I appreciate any help I can get!

Thanks!

cyberman54, 2021-03-23:
Please, if you request support, use Discussion tab in Github, do not open issue, thanks.

Remote control by rcommands using TTN console is pretty easy: Go to the downlink menu (see screenshot), switch to FPORT2 and enter the bytes of rcommand (see readme.md for command list) in HEX, then press send. The device will receive and execute the command after next handshake with TTN, that is, wenn it sends next payload. You can trigger this by press and hold the button of your device (if it has user button configured).

[image]

closed 2021-03-23 by cyberman54

SionHuws99, 2021-03-23:
Thank you ! Sorry for posting in the wrong place.

Quick question, so for example 0x04 is change display and 0 = display off. How is this then entered in the downlink?

Sorry if this is a silly question!