cyberman54/ESP32-Paxcounter #776: WIFI_SSID and WIFI_PASS not declared

sinalco92, 2021-03-30:
I become a compiling Error in the "src/boot.cpp" data, because WIFI_SSID and WIFI_PASS are not declared.

`src/boot.cpp: In function 'void start_boot_menu()':
src/boot.cpp:88:22: error: 'WIFI_SSID' was not declared in this scope
   const char *ssid = WIFI_SSID;
                      ^
src/boot.cpp:89:26: error: 'WIFI_PASS' was not declared in this scope
   const char *password = WIFI_PASS;`

cyberman54, 2021-03-31:
Did you create ota.conf?

sinalco92, 2021-03-31:
> Did you create ota.conf?

Yes I change the ota.sample.conf to ota.conf
But I dosen’t change anything in the ota.conf

cyberman54, 2021-03-31:
I can't reproduce this. Maybe an issue with your toolchain?
The defines `WIFI_SSID `and `WIFI_PASS `are created dynamically in buildscript `build.py`.
Please check if this works on your platform.

sinalco92, 2021-03-31:
> I can't reproduce this. Maybe an issue with your toolchain?
> The defines `WIFI_SSID `and `WIFI_PASS `are created dynamically in buildscript `build.py`.
> Please check if this works on your platform.

Ok now it works.
The `build.py` was not in the project folder. 
Sorry for that, it was my mistake
Thanks for supporting!

closed 2021-03-31 by cyberman54